dronekit/dronekit-python #398: Allow the data stream rate to be set by the API

hamishwillee, 2015-11-03:
From gitter (@a7scenario)

> Does DK2 have a way to "permanently” change the rate at with the autopilot streams a specific message type (or set of msg types in a MAV_DATA_STREAM set)? For whatever reason, sending a request_data_stream message only works for a couple seconds and then the autopilot reverts back to the previous rate.

   ```
msg = self.vehicle.message_factory.request_data_stream_encode(0, 0, mavutil.mavlink.MAV_DATA_STREAM_EXTRA3, new_rate, 1)
self.vehicle.send_mavlink(msg)
self.vehicle.flush()
```

squilter, 2015-11-03:
Setting the parameter to 10 will leave the parameter at 10.  The problem is that ardupilot seems to stream the parameter at 10Hz at first, but then slows down.  I've seen this many times, and I think it's caused by ardupilot.

eliao, 2015-11-03:
This is probably because many GCSes set their own rate every few seconds, which clobbers the 10 hz.  For example, Mission Planner, mavproxy, and I believe Tower all set some rates to 4 hz.  Mavproxy has an option to stop clobbering stream rates:  --streamrate=-1

squilter, 2016-01-12:
I think this can be closed.  It works for me when I use `dronekit.connect(path, rate=20) `

a7scenario, 2016-01-12:
@squilter  Did the rate change "stick" while running your GCS?

squilter, 2016-01-12:
yes, it worked when I just tried it.

hamishwillee, 2016-01-20:
@squilter @a7scenario As you [can see here](https://github.com/squilter/dronekit-python/blob/versions_and_capability/dronekit/__init__.py#L2011), the rate is set only during initialisation. That means that if the rate is changed later by the GCS on the same channel, then DroneKit is doing nothing about that.

If you're seeing this stick then either the the GCS and dronekit are on a different mavlink channels or the GCS you're using is not trying to set the rate (some continually reset it, and others do not).

I do not particularly have a problem with us having an API to set rate, but it really should be considered a "desired rate" given that other systems on the channel can change it. 

What I would recommend is that if your app does depend on a certain rate, observing the [SRX_](http://copter.ardupilot.com/wiki/arducopter-parameters/#sr0__parameters) parameters to monitor that the rate is OK. At least then you have more debug on the problem configurations.