dronekit/dronekit-python #400: send_mavlink() doesn't autofix system id

juanmacrack93, 2015-11-03:
When sending mavlink messages created with message_factory, the correct system id must be specified, otherwise it won't work.

This was tested using the following code:

```python
 msg = vehicle.message_factory.mount_control_encode(
        0, 0,  # target system, target component
        30 * 100,  # pitch
        0,  # roll
        0,  # yaw
        0)  # save trim
```

However, changing the system id to the proper value (1 in this case) fixes the problem.
The documentation suggests that this should be automatically taken care of by send_mavlink(), yet it does't seem to be the case.

I tried the same code with dronekit 1 and it does indeed work without the need to specify the sysid manually.

tcr3dr, 2015-11-03:
Hi @juanmacrack93,

Thanks for filing the issue. Message correction is added back in with the refactor on this PR: https://github.com/dronekit/dronekit-python/pull/401

Exposing a way to change the target system and component is not on that PR yet, but should be added in shortly.

tcr3dr, 2015-11-11:
You can follow #448 for a way to modify the system/component directly.

tcr3dr, 2015-11-18:
Further addressed by #461.

tcr3dr, 2015-11-18:
Merged.

closed 2015-11-18 by tcr3dr