dronekit/dronekit-python #412: HDOP incorrectly represented in cm in the Documentation

jainakshay91, 2015-11-04:
Hi, is the HDOP value returned by dronekit-python from gps_0.eph actually in centimeters ? As much as I remember, a DOP value is just a factor and is a unitless quantity. Further, most GPS receivers supply information in the NMEA format. Now the GPGGA message in the NMEA protocol which provides the HDOP information says that there is no unit to this quantity (reference: ublox protocol manual page 68). Next, since the AP_GPS_NMEA.cpp is used to precisely parse these NMEA messages and then pack them into the mavlink protocol messages, I cannot find a step where these HDOP (unitless) values are converted to cm. The function to parse these messgaes:_parse_decimal_100() is a generic function which is also used for parsing the time and altitude information. In the definition of _parse_decimal_100() it states that there is a theoretical resolution of upto 1 cm and hence I think there is this confusion with the units of the HDOP value.  Kindly let me know thoughts and comments on this issue. 

cc: @hamishwillee

hamishwillee, 2015-11-04:
My understanding of HDOP matches yours - its a unit-less ratio for determining how accurate your position is likely to be.

I can see that [when it comes in we just copy it from the NMEA string](https://github.com/diydrones/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_NMEA.cpp#L310). It does look like some scaling is going on in [the driver here](https://github.com/diydrones/ardupilot/blob/3d7ec3704d88053ae03b6e01077946ec2084af99/libraries/AP_GPS/AP_GPS_PX4.cpp#L58). 

Not sure, I'll ask the experts in ArduPilot land.

jainakshay91, 2015-11-04:
Okay. Do you want me to raise an issue on the ardupilot github forum and then tag you there? That way we can chase them over this issue.

billbonney, 2015-11-04:
HDOP is correctly represent in the documentation as the HDOP is sent as part of the GPS_RAW message from MAVLink, which sends it in cm.
The issue maybe that the ArduPilot FW is sending it incorrectly, so raise the issue here https://github.com/diydrones/ardupilot

jainakshay91, 2015-11-04:
Well, the interpretation of HDOP isn't correct throughout that is what I meant, and so I still do not think that it is correct in the documentation even though the GPS_RAW message says the eph is in cm. The GPS_RAW message is just a protocol to pack the raw GPS information, for which the original information is drawn from the NMEA message being sent by the GPS module. And if we look at the NMEA message protocol the HDOP information is unitless. @hamishwillee did mention about some scaling being carried out in a driver. So I am looking into that. 

I do agree with you and Hamish that this is an ArduPilot issue, and so I will go raise this issue on the Ardupilot forum as well. Thanks so much and let me know if any new information comes up regarding this.

billbonney, 2015-11-04:
Well the protocol is a standard, that other autopilots are using like PX4 Flight Stack, ArDrone Native Stack, AeroQuad etc.. The protocol can be changed, but then you need many more places to effect commonality. From a GCS point-of-view variations are going to be a pain. http://mavlink.org for protocol related stuff

billbonney, 2015-11-04:
BTW: The ArduPilot and PX4 autopilots use the UBLOX binary protocol, not the NMEA strings (I don't think anybody uses NMEA has its update rate is too slow cumbersome on a microconotroller) so you need to look at the UBLOX drover to understand the hoop. Also it was changed recent, search fro hoop at drones-discuss.googlegroups.com

hamishwillee, 2015-11-04:
@jainakshay91 ( and @billbonney )

So this is a bit of a mess, but I think I'm nearing the bottom of the rabbit hole :-).

The spec is basically wrong for DOP because it is a unitless property. The implementation of what is returned by this message is different on PX4 and ArduPilot:

* PX4 - Abuses the value - it returns hacc and vacc (instead of hdop and vdop, respectively). These values arrive in mm and are converted to metres (in scaled integers). 
* Ardupilot - returns  the uBlox hDOP for uBlox devices that support the MSG_DOP uBlox message (and similar for vDOP). So this is the unscaled unitless value as you would expect.

I am still chatting to the guys to see if we can "rationalise" this ... after which I'll look at the dronekit docs.

jainakshay91, 2015-11-04:
That's great to hear. Is there anyway that I can help you guys out in sorting this mess?

hamishwillee, 2015-11-04:
Nope. Looks like PX4 will just change its implementation to report "the right thing" and we'll  update the spec to remove the units. I'll keep you posted.

jainakshay91, 2015-11-04:
Thanks a lot @hamishwillee . That is great to hear again.

hamishwillee, 2015-11-04:
DK docs are fixed in #451. I have put in request for change to the protocol definition in https://github.com/mavlink/mavlink/pull/451

billbonney, 2015-11-05:
But without specifying what the unitles scaler means, the value is worthless. Is this an accurate representation of the HDOP? https://en.wikipedia.org/wiki/Dilution_of_precision_(GPS)#Meaning_of_DOP_Values
And is this going to be common across devices? at least when it's converted (or should be converted) to a unit it has some comparison that can be assume in a GCS, for example

hamishwillee, 2015-11-05:
HDOP/VDOP is understood to mean something to people who use GPS. AFAIK that list makes sense as a means of interpretation, but I'm not a GPS expert.

As far as I can tell it *is* pretty useless for UI purposes- you're much better off putting out a standard deviation or similar, which actually means something to users (ie can be drawn as a circle around current point to show you're "probably here"). 

I have no idea if all manufacturers implement the spec in the same way. What I do know is that the previous wording was wrong, and the current wording is correct - though you might argue it is not particularly useful.

billbonney, 2015-11-05:
We do present HDOP in the current UI as the docs are indications what the value should be. see http://copter.ardupilot.com/wiki/prearm_safety_check/#failure_messages (I know you know ;) )
BUT it would be nice to show a circle of precision. Changing the MAV Link protocol spec might actually not be the right thing here. (or might not even be accepted) Probably good to ask this on drone-discuss

hamishwillee, 2015-11-05:
Ah, according to Tridge we're not changing anything - just the documentation (this is what is output from ArduPilot on GPS_RAW_INT. BUT on PX4 stack  is actually a standard deviation in metres. That's what you get when you have a spec that makes no sense.

So what might change is that PX4 will change the meaning to "really" be HDOP.

The right thing to do in the UI is to get hold of the standard deviation, which you can display. Yes, worth a discussion in DIYDRONES

jainakshay91, 2015-11-05:
So, first thing first what is HDOP. HDOP shows how good the geometry of the satellites is over the GPS receiver so as to be able to provide a unique solution to the problem of determining the GPS time as well as the (x,y,z) coordinates of the receiver. If the geometry is not good (satellites are collinear or concyclic with the receiver) then you get high HDOP values. 

Further,  representing a circle of precision would require us to estimate the UERE which I think is provided in the hacc value of one of the UBlox messages. So, maybe using the hacc you can draw a circle of precision.

jainakshay91, 2015-11-05:
So @hamishwillee can you summarize the changes, as I am getting a little lost here. Sorry for the trouble.

hamishwillee, 2015-11-05:
Proposed changes are summarised in #418 and #415 above (ie these are the precise PRs). They are just documentation changes to remove the units from DK and protocol documentation. These will then reflect what ArduPilot does. They will not reflect what PX4 returns.

billbonney, 2015-11-05:
Quite interestingly the value is marked as eph and epv in the message. see http://forums.enterprisemobiletoday.com/showthread.php?40852-Can-someone-explain-these-GPS-concepts So when the message was created the intent was obviously different.
MAybe we can rename the param names in a future release if hdop is standardized in the standard ;-).

closed 2015-11-06 by tcr3dr

hamishwillee, 2015-11-06:
@billbonney Absolutely. Personally I'd prefer that the field was eph  (as per that definition) as it is far more useful. That would require ArduPilot change its implementation and the docs be updated. The former is outside of scope for a Dronekit issue :-)

PR #418 (2015-11-06): Fix GPSInfo to remove units from eph and epv definition
This fixes #412 by removing the units from the description of epv/eph. These units were copied from the MAVLINK spec for [GPS_RAW_INT](https://pixhawk.ethz.ch/mavlink/#GPS_RAW_INT) - but the spec is "wrong" because the value is unitless ... and I have confirmed that ArduPilot does return an unscaled HDOP/VDOP value.