dronekit/dronekit-python #413: Update examples to reflect changes to observer methods

franknitty69, 2015-11-04:
As of DK 2.0.0 rc7, examples that utilize observer methods will fail because they are still using the old syntax and logic.

To fix the the issue, all examples using observer methods need the following changes:

1. call back definitions require 2 parameters, not 1
replace **def mode_callback(attribute)** with **def mode_callback(vehicle, attr_name)**

2. replace **add_attribute_observer** with **on_attribute**
3. replace **remove_attribute_observer** with **remove_attribute_listener**

Examples on python.dronekit.io need to be updated as well.

tcr3dr, 2015-11-04:
Will have the docs updated shortly. Also adding this list to our migration guide #379

hamishwillee, 2015-11-05:
#420 starts this task. Still working on it. @franknitty69 we are likely to change the names of some of the methods too.

tcr3dr, 2015-11-07:
Renaming these methods: #426

tcr3dr, 2015-11-09:
#420 and #426 have been merged. rc9 has these changes and the python.dronekit.io docs have been updated accordingly. What remains is the migration guide at #379.

I believe we can move discussion to those issues. Thanks!

closed 2015-11-09 by tcr3dr