dronekit/dronekit-python #414: No timeout on "connect"

mikerob, 2015-11-04:
It appears that there's no timeout for the "connect" function.  If you try to connect to a serial port that exists but is not actually connected to a vehicle, it just hangs.

On the RasPi, I'm conecting to the Pixhawk via USB, so it's /dev/ttyACM0. The RasPi has a built in serial port at ttyAMA0. If you "Connect('/dev/ttyAM0')" it hangs, presumably because it's a valid port so it thinks it can read, but never gets valid data.

mikerob, 2015-11-04:
In dronekit-python / dronekit / lib / __init__.py it calls "wait_heartbeat", which doesn't timeout (mavutil.py - lines 326 and 372)

I'm not sure what the solution is, other than to write a new "wait_heartbeat_timeout" function that calls recv_match from mavutil and passes it a timeout value, rather than using the ready-made wait_heartbeat.

# Wait for first heartbeat.
        while True:
            try:
                self._master.wait_heartbeat()
                break
            except mavutil.mavlink.MAVError:
                continue

tcr3dr, 2015-11-04:
@mikerob,

Thanks for pointing this out. I think this can tie pretty well into creating actual connection timeouts (rather than intermittent dropoffs), which there is a branch exploring now: https://github.com/dronekit/dronekit-python/pull/407

The same connection logic which says after 30s, close the connection and throw, can be used on initialization for the same purpose.

That branch is a WIP but I want to land it soon, as well as making these timeouts (and their verbosity) configurable.

mikerob, 2015-11-04:
Sounds like there's work already underway.  I'll take the liberty of running through what I'm doing with this, because I have a number of issues - some of them may be relevant to a wider audience.

Overall, I'm working on a lightweight ground station based on a RasPi and 20x4 character LCD (a bit like an updated ArduStation).  I'd like it to be flexible and hands-free (and headless) in terms of connection, so ideally could "automatically" detect how a vehicle is connected.  To that end, I'd like to be able to try network, USB radio, USB Pixhawk, and direct GPIO serial connections.  

1) Managed / more graceful handling of connections that don't exist - would be nice to have the connection return a softer error saying the IP or serial port does not exist. This would make it easier to try different connections.

2) The primary issue rasied above - valid serial port, but no incoming data

3) Managing loss of connection. As I see it there are two different loss of connetion cases:
    a) Local end remains connected, but remote vehicle drops off - this is the "normal" case where you stop receiving heartbeat packets. The system seems to already handle waiting for data to start flowing again.
   b) Loss of local end - if USB cable gets unplugged at the Pi, or the radio gets powered down, or network cable comes out - your actual connection at the Pi end becomes invalid.  The biggest problem currently with this case (and maybe it's only on the Pi) is that when you reconnect the radio or Pixhawk, it comes back as a different serial port - it goes from /dev/ttyUSB0 to /dev/ttyUSB1.

    It woudl be good to detect and fail gracefully (as with the initial complete connection fail) if the serial port you're accessing falls over.

  30s is a long time for a timeout - seems like it would be suitable as a default, but would be nice to be able to customise.

  On the overall connection timeout, it seems one way to handle / report it would be to add a Vehicle.heartbeat parameter that is initialised to -1 and otherwise records the time since the last heartbeat packet was received. That would make it easy to check if a connect failed (heartbeat = -1) and maybe could be set to -1 as well if the local end connection fails.

  If I have a 3DR radio connected, does that send RADIO messages even when no vehicle is connected?

  I'm happy to help test the connection stuff if you like.

tcr3dr, 2015-11-11:
Hi @mikerob,

As of #447, I think several of your points are now addressed. This should demonstrate the behaviors you are expecting:

```
import dronekit
import socket
import exceptions

# Bad TCP connection
try:
	dronekit.connect('tcp:127.0.0.1:8888')
except socket.error:
	print 'No server exists!'
except dronekit.APIException:
	print 'Timeout!'
except:
	print 'Some other error!'
	
# Bad TTY connection
# tested using: sudo socat udp4:127.0.0.1:5998 pty,link=/dev/ttymxc2,echo=0,raw,nonblock,user=timryan
try:
	dronekit.connect('/dev/ttymxc2', heartbeat_timeout=5)
except exceptions.OSError as e:
	print e
	print 'No serial exists!'
except dronekit.APIException:
	print 'Timeout!'
except:
	print 'Some API error!'
```

`heartbeat_timeout` is as documented on http://python.dronekit.io/. Also you may still have to import `dronekit.lib.APIException` but this is temporary.

Let me know what else we should address; I would like to get a release-quality DKPY2 version out soon but we need most backward-incompatible changes in place first.

tcr3dr, 2015-11-11:
Note this is now merged into master.

mikerob, 2015-11-12:
Darn, spent the evening testing it with my own code because I hadn't seen the code you posted yesterday.

Happy to consider closed.

closed 2015-11-17 by tcr3dr