dronekit/dronekit-python #415: Channel Overrides Error - Vehicle object has no attribute '_Vehicle__rc'

franknitty69, 2015-11-04:
When running the example, Channel Overrides, the script fails.

OS: OSX
Dronekit: 2.0.0rc7
channel_ovverides.py: 2015/11/04

**steps to reproduce**
1. python channel_overrides.py --connect=127.0.0.1:14550

**result**
Overriding RC channels for roll and yaw
Traceback (most recent call last):
  File "/Users/nyce/Documents/GitHub/dronekit-python/examples/channel_overrides/channel_overrides.py", line 35, in <module>
    vehicle.channel_override = { "1" : 900, "4" : 1000 }
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/dronekit/lib/__init__.py", line 993, in channel_override
    overrides = self.__rc.override
AttributeError: 'Vehicle' object has no attribute '_Vehicle__rc'

**expected results**
Current overrides and defaults should be displayed

tcr3dr, 2015-11-04:
Thanks for filing this issue @franknitty69. Related: #358 

RC support isn't in rc7, but should be reimplemented shortly in another rc.

tcr3dr, 2015-11-06:
See #427.

tcr3dr, 2015-11-11:
Fixed as of #427.

closed 2015-11-11 by tcr3dr