dronekit/dronekit-python #416: Waiting on arming

tcr3dr, 2015-11-04:
Considering the GPS and EKF issues exposed by #406, I wonder if we shouldn't be exposing a different parameter indicating "readiness to arm".

In several of our examples, we sidestep this by waiting for a GPS lock. e.g. in simple_goto:

```
    print "Basic pre-arm checks"
    # Don't let the user try to fly autopilot is booting
    if vehicle.mode.name == "INITIALISING":
        print "Waiting for vehicle to initialise"
        time.sleep(1)
    while vehicle.gps_0.fix_type < 2:
        print "Waiting for GPS...:", vehicle.gps_0.fix_type
        time.sleep(1)
```

We discover with 3.3 we also need to wait for an EKF readiness flag. What if we were instead to abstract several of these considerations into an attribute, `vehicle.armable`:

```
vehicle = connect('...', wait_ready='armable')
vehicle.armed = True
```

This would encompass the MODE, GPS, and (depending on vehicle?) EKF checks, so that we can have a simple mechanism to wait for this (instead of our crufty spin-loops).

tcr3dr, 2015-11-04:
We may also want to expand the `wait_ready` mechanism to encompass matching values.

```
vehicle = connect('...', wait_ready='armable')
vehicle.armed = True
vehicle.wait_condition(armed=True)
```

mrpollo, 2015-11-04:
what if i have a special condition they would like to meet, can i extend armable?

tcr3dr, 2015-11-04:
(I actually take back my earlier comments about extending `wait_ready` and `wait_condition`; that would be mixing information synchronization with conditions. Conditional checks should probably remain in user code.)

tcr3dr, 2015-11-04:
@mrpollo Perhaps, to keep this simple, we can go for something like the following:

```
class Vehicle():
    ...
    @property
    def is_armable(self):
        return GPS and EKF and MODE
    ...
```

And subclasses of Vehicle (using the vehicle_class mechansim) can just override is_armable(). Thus our code for simple_goto becomes:

```
print "Basic pre-arm checks"
while not vehicle.is_armable:
    time.sleep(1)
vehicle.armed = True
while not vehicle.armed:
   time.sleep(1)
```

This is still ugly, but we don't have any general solution yet for "await condition in user code with timeout" and that sounds like a larger, separate issue.

mrpollo, 2015-11-04:
agreed

hamishwillee, 2015-11-04:
@tcr3dr We can't really wait on armable anyway on connecting, because some states on some vehicles don't allow arming. 

I like the idea of isarmable as a check. I lovethe idea of being able to get the reason it isn't armable back from the system (though I'd really only be interested on getting the value(s) holding it back). This is important because if it doesn't arm I need to easily be able to work out why. 

In example, don't both True and ``return MODE and GPS and (if we have EKF, EKF)`` both evaluate as true?

It would be great if we could fix the await condition in user code with timeout issue - but what you've got above is workable. Makes it much easier to build a straighforward "takeoff" function, which is what I think users really want :-)

tcr3dr, 2015-11-11:
The goal of this issue, having cross-platform code that supports waiting for arming, is implemented by #430. The mechanism by which we improve on this in the future remains to be seen, but for now I'm closing this issue.

closed 2015-11-11 by tcr3dr

hamishwillee, 2015-11-11:
Agreed. #430 is pragmatic, and I think using it in anger for a bit may tell us that it is "good enough".