dronekit/dronekit-python #417: ImportError: No module named dronekit on 2.0.0rc7

kevindgoff, 2015-11-04:
Sorry if this is wrong place for this but I'm new to DK.  

I've been following the instructions on the getting started setup.

I'm getting the following error on MacOS (el capitan). I'm out of ideas on how to fix. i've installed, uninstalled multiple times.

Traceback (most recent call last):
  File "/Users/kgoff/Documents/dronekit/dronekit-python/examples/vehicle_state/vehicle_state.py", line 9, in <module>
    from dronekit import connect
ImportError: No module named dronekit

pip list output:
dronekit (2.0.0rc7)
dronekit-sitl (3.0.1)
MAVProxy (1.4.37)
pip (7.1.2)
psutil (3.2.2)
pymavlink (1.1.62)
pyserial (2.7)
requests (2.5.1)
setuptools (12.0.5)
vboxapi (1.0)

tcr3dr, 2015-11-04:
@kevindgoff Happy to help. I've run into these paths issues a few times as well.

Can you run and paste the result of these two commands:

```
pip show -f dronekit | grep Location
```

and

```
python -c "import sys; print(sys.path)"
```

Thanks!

hamishwillee, 2015-11-04:
@kevindgoff The last two times someone reported this it was because they had multiple versions of python installed - and the version that they were running the scripts on was not the same version that was being updated by PIP. For example, I was getting different invocations of Python by doing ``python myscript.py`` vs just ``myscript.py``. Worth checking.

If not, @tcr3dr debugging information above is your next check!

kevindgoff, 2015-11-05:
Thanks to both of you for quick response...
I am using python "myscript.py" to call it as I get a bash error just called the file direct.  

Here is the output from the above commands:
GCI-KGOFF2-M:dronekit-python kgoff$ pip show -f dronekit | grep Location
Location: /usr/local/lib/python2.7/site-packages
GCI-KGOFF2-M:dronekit-python kgoff$ python -c "import sys; print(sys.path)"
['', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python27.zip', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-scriptpackages', '/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-old', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload', '/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC', '/Library/Python/2.7/site-packages']

kevindgoff, 2015-11-05:
As i look in my /usr/bin folder I am seeing several symlinks and files related to python 2.6.  I'm attempting to safely remove those links. I'll report back soon...

tcr3dr, 2015-11-05:
Yeah, it looks like `/usr/local/lib/python2.7/site-packages` is not in the system path for Python. Did you happen to install python via `brew`?

Also, you can try running `python -m pip install dronekit` and then importing. That uses the pip version for the specific version of Python that seems to have higher precedence.

kevindgoff, 2015-11-05:
This has been worked out and was related to python install.  I struggled through it a bit, almost hosed python altogether, but ultimately i think it might have just been a simple addition to the path variable to make sure it was getting it from /usr/local/bin/python vs.  /usr/bin/python.

thanks for the pointer in the right direction. looking forward to what can be done with this toolkit!

closed 2015-11-05 by kevindgoff

tcr3dr, 2015-11-06:
@kevindgoff Glad to hear it! We can now help other users debug issues like this, which is good.

hamishwillee, 2015-11-06:
Yes - thanks! @tcr3dr While it is tempting to add a troubleshooting section to the docs I suspect google will pick this up this issue just as easily if people have a problem.