dronekit/dronekit-python #421: Connection error with 3DR radio

aknight0, 2015-11-05:
I'm trying to connect Dronekit python to my Pixhawk over the 3DR radio set.  I tested first with Mission planner, and was able to connect just fine.  However, when I disconnect Mission Planner, and attempt to connect with Dronekit, I get the following error after a long wait:

    Connecting to vehicle on: com3
    Traceback (most recent call last):
      File "vehicle_state.py", line 27, in <module>
        vehicle = connect('com3', await_params=True)
      File "C:\Users\UAV-PRISM\Anaconda\lib\site-packages\dronekit\__init__.py", line 586, in connect
        state.prepare(await_params=await_params, rate=rate)
      File "C:\Users\UAV-PRISM\Anaconda\lib\site-packages\dronekit\__init__.py", line 548, in prepare
        self.master.wait_heartbeat()
      File "C:\Users\UAV-PRISM\Anaconda\lib\site-packages\pymavlink\mavutil.py", line 374, in wait_heartbeat
        return self.recv_match(type='HEARTBEAT', blocking=blocking)
      File "C:\Users\UAV-PRISM\Anaconda\lib\site-packages\pymavlink\mavutil.py", line 339, in recv_match
        m = self.recv_msg()
      File "C:\Users\UAV-PRISM\Anaconda\lib\site-packages\pymavlink\mavutil.py", line 313, in recv_msg
        msg = self.mav.parse_char(s)
      File "C:\Users\UAV-PRISM\Anaconda\lib\site-packages\pymavlink\dialects\v10\ardupilotmega.py", line 7962, in parse_char
        m = self.__parse_char_legacy()
      File "C:\Users\UAV-PRISM\Anaconda\lib\site- packages\pymavlink\dialects\v10\ardupilotmega.py", line 7988, in __parse_char_legacy
        (magic, self.expected_length) = struct.unpack('BB', str(self.buf[0:2])) # bytearrays are not supported in py 2.7.3
    struct.error: unpack requires a string argument of length 2

Any ideas?  I also tried connecting to a SITL instance running on the same computer, and that worked just fine.

hamishwillee, 2015-11-05:
I personally don't have any idea, but can you let us know what build of DroneKit you're using?

tcr3dr, 2015-11-06:
@aknight0 This may be a hard issue to diagnose, but can you try running this on master by installing via pip:

```
pip install -UI git+https://github.com/dronekit/dronekit-python
```

We are no longer using the `wait_heartbeat` function, so this may work better.

aknight0, 2015-11-06:
Upgrading to the newest master seems to have helped, thanks!  I also had to set baud=57600 to get a successful connection, the default setting always times out.  This is with dronekit-2.0.0rc7 and APM 3.3.1 on the Pixhawk.

tcr3dr, 2015-11-07:
@aknight0 This is good to hear! Our default baud is fairly high so that makes sense.

I'll be closing this issue if that's okay. I think this is related to #263 ultimately so we'll probably investigate the root cause in the future.

closed 2015-11-07 by tcr3dr