dronekit/dronekit-python #432: Setting connection timeout too short (<10s) throws unhandled exception

mikerob, 2015-11-07:
I was testing out the connection timeout, and came across an odd behaviour.  If I set the timeout > 10 seconds, everything works nicely - it hapily throws an APIException that I can catch.

However, if I set the connection timeout lower than that, it throws a different exception, seemingly from somewhere "inside" the connection, and it's an "Exception" rather than an "APIException".

Is this the intent?  I reported it here: https://github.com/dronekit/dronekit-python/pull/407 and was requested to submit it as a new issue.

from dronekit import connect
 from dronekit.lib import APIException

try:
 vehicle=connect('/dev/ttyAMA0', heartbeat_timeout=10)
 except APIException as ai:
 print "APIException fail - Something bad happened"
 print type(ai)
 print ai.args
 print ai.msg
 except Exception as ai:
 print "Something we didn't expect"
 print type(ai)
 print ai.args
 print ai
 else:
 print "It worked"

If I set timeout to 10 or more, it works fine. If less than that, it throws the following:
 pi@MavStation ~/pigs $ python to_exc.py

 

Link timeout, no heartbeat in last 5 seconds
 Exception in thread Thread-1:
 Traceback (most recent call last):
 File "/usr/lib/python2.7/threading.py", line 552, in bootstrap_inner
 self.run()
 File "/usr/lib/python2.7/threading.py", line 505, in run
 self.__target(self.__args, *self.__kwargs)
 File "/usr/local/lib/python2.7/dist-packages/dronekit-2.0.0rc7-py2.7.egg/drone
 kit/__init.py", line 77, in mavlink_thread
 fn(self)
 File "/usr/local/lib/python2.7/dist-packages/dronekit-2.0.0rc7-py2.7.egg/drone
 kit/lib/init.py", line 855, in listener
 raise Exception('>>> No heartbeat in %s seconds, aborting.' % self._heartbea
 t_error)
 Exception: >>> No heartbeat in 9 seconds, aborting.
 
APIException fail - Something bad happened
 
()
 Timeout in initializing connection.

It looks like setting timeout too low interferes with or gets inside some other exception handling loop.

tcr3dr, 2015-11-09:
Thanks for filing this, @mikerob. This is definitely a weird case for exceptions—one being throw on the MAVLink thread, another interrupting the main thread—so let me put together a proposed solution for this that is catchable.

hamishwillee, 2015-11-11:
Hi @mikerob 

YOu might want to try out the branch in #447, which @tcr3dr thinks improves this issue.

tcr3dr, 2015-11-11:
Note this is now on master.

mikerob, 2015-11-12:
@tcr3dr Thanks Tim - looks like it's working now. Do I "close" this, or do you?

I've got some test code in here: https://github.com/mikerob/dronekit-python/tree/master/pigs if you're interested

tcr3dr, 2015-11-17:
I can close it. Thanks for the test code! This is helpful.

closed 2015-11-17 by tcr3dr