dronekit/dronekit-python #433: Allow Dronekit to output MAVLink stream similar to MAVPRoxy

mikerob, 2015-11-07:
I'm developing a lightweight ground station, based on a RasPi and LCD display.  The idea is that it provides very basic info - GPS location, heading, altitude and RSSI on a daylight readable display.

Under DroneKit 1, it ran inside a MAVProxy instance set up to read from the radio (/dev/ttyUSB0) and send out to udpout for UDPCI connections from a PC-based ground station.

Now with DKPY2, I have to run a MAVProxy with an additional --out=udp:127.0.0.1:14551 and hook into that from a separate Dronekit script.

Is it feasible to have DKPY2 provide the same stream output functionality as MAVProxy?

hamishwillee, 2015-11-08:
@mikerob I can't comment on the implications of this (I don't fully understand the problem). My gut feeling though would be that it doesn't make sense for dronekit to reinvent the wheel and try to reimplement what MAVProxy does best. 

@tcr3dr @mrpollo One for you to comment on.

tcr3dr, 2015-11-08:
@mikerob Agreeing with Hamish here, the best solution is probably continuing to use MAVProxy. It is designed for this usecase.

I see the value in integrating MAVLink forwarding components into DKPY2 though. Probably not for the upcoming release, but we need a similar component in dronekit-sitl so we'll just track progress here.

mikerob, 2015-11-09:
@hamishwillee, without this functionality, DKPY2 is, in some respects, a retrograde step compared to DK1. There are a number of use cases for for the forwarding that don't utilise *any* of the other functionality of MAVProxy, so you're loading up your companion (or other) computer for nothing if you have to run both DKPY2 and MAVProxy.

1) My case - lightweight ground indicator with forwarding to tablet / PC. The only functionality being useed from MAVProxy is the forwarding, but MAVproxy (AFAIK) doesn't handle connection failures well - a USB-based reconnect that increments your ttyUSBx will not recover, and you can't build in any smart logic for detecting baud rate mismatches. All those things are feasible with DKPY2 ... except for the forwarding.

2) Wifi / cellular / other telemetry - if you want to pass your data over anything other than a serial port connection from the on-board autopilot, you need to run it through a companion computer. This requires the whole of MAVProxy just to pass it from a serial input on the Pi to a USB or network interface. If you want to concurrently use and DK functionality, you have to do the output doubling from MAVProxy.

tcr3dr, 2015-11-09:
@mikerob These might be issues you want to file with MAVProxy directly. That tool is designed for the usecase of being a general-purpose MAVLink forwarder, and it would be equivalent effort to implement it in DKPY2 as it would be to fix MAVProxy to suit your usecases.

That said, if you were to begin implementing code that uses pymavlink directly to implement connection forwarding, it is a strong candidate for inclusion in DKPY2. Conceptually it seems straightforward, and you have the benefit of ensuring it works with your setup first and foremost.

Aside: The biggest reason for moving away from MAVProxy as a base is that code written in DKPY1 was not thread-safe or easily debuggable. This was only an artifact of how DKPY1 was designed though, not any issue with MAVProxy afaik.

pietrodn, 2018-04-01:
In #812 I posted how I got to forward a MAVLink stream from a serial port to UDP using DroneKit. Is this sufficient?