dronekit/dronekit-python #455: API explanations in the top level Vehicle or convenience classes?

hamishwillee, 2015-11-16:
If you have an attribute like ``vehicle.heading`` which represents a single value there is just one place where it can be documented.

For a compound object like ``Vehicle.location`` you can have the documentation in the top level attribute or spread in the sub classes like ``Locations``, ``LocationGlobal``, ``LocationGlobalRelative`` etc. 

Having most of the docs up in the top level ``vehicle`` makes sense because you mostly operate on the attributes through this object, but it does mean that you can't easily scan through ``Vehicle`` to see the available attributes and then dig down for functionality. This wouldn't matter so much if there was a class listing in the sidebar (as requested) but it is a bit of a pain as we don't.

Most of all though, it is important to try be consistent. My leaning is to have most of the documents in the top level ``Vehicle``, with minimal documentation in "sub attributes" (and hope our docs infrastructure improves).

@tcr3dr @mrpollo Do you have any opinion on this?

tcr3dr, 2015-11-16:
Since this is a UX issue we're solving through hacks, I would lean toward your proposed solution. We should improve the structure of our docs, but take the easiest path now (even if it's not correct).

mrpollo, 2015-11-16:
I think we should focus on supporting the 2.0 release in any way we can, we can fix the docs afterwards

hamishwillee, 2015-11-16:
Thanks. The api reference is OK and is well supported in the guide. We can do a cleanup for consistency later as you (both) suggest.

closed 2015-11-16 by hamishwillee