dronekit/dronekit-python #463: how to send values for copter and plane attitude(pitch,roll,yaw), if i cant use chanel_override?

ianugrohocux, 2015-11-18:
i have two projects that use computer vision(OpenCV) and dronekit python (plane and copter). the projects are how to make plane(fly wing) to fly on autonomous mode that track/follow river use the input from computer vision, and how to make quadcopter fly also on autonomous mode that follow the movement of circle object(left/right foward/backward). i use APM 2.6 for my flight controller. i have tried channel_override and i got confused to analyze the plane and copter attitude. the main task is how to give values to plane and copter attitude (yaw,pitch,roll,altitude) from the computer vision feedback.

tcr3dr, 2015-11-18:
Hi @ianugrohocux,

This question would be better suited for the [discussion forums](https://discuss.dronekit.io/). If you could re-post your question there we'd appreciate it. Thanks!

closed 2015-11-18 by tcr3dr

ianugrohocux, 2015-11-19:
oke thanks @tcr3dr