dronekit/dronekit-python #467: Vehicle.commands.goto always treats alt as relative

hamishwillee, 2015-11-19:
Vehicle.commands.goto takes a LocationGlobal or a LocationGlobalRelative but either way it assumes the alt value is relative. This means that if you pass it a LocationGlobal and say "goto" it will fly to that position ABOVE the home position (rather than mean sea level). 

Internally we need to detect we're dealing with a global and pass in the associated relative alt.

tcr3dr, 2015-11-19:
@hamishwillee I'm not sure I'm seeing that's the case. See this:

```
import time
from dronekit import connect, VehicleMode, LocationGlobal, LocationGlobalRelative

v = connect('tcp:127.0.0.1:5760', wait_ready=True)

print 'await armable'
while not v.is_armable:
	time.sleep(1)

print 'await guided'
v.mode = VehicleMode('GUIDED')
while v.mode != 'GUIDED':
	time.sleep(1)

print 'await armed'
v.armed = True
while v.armed != True:
	time.sleep(1)

print 'takeoff'
v.commands.takeoff(10)

time.sleep(5)

print 'okay'
print v.location.global_frame

v.commands.goto(LocationGlobal(v.location.global_frame.lat, v.location.global_frame.lon, v.location.global_frame.alt + 20))

print 'going to global frame...'
time.sleep(10)

print 'global frame', v.location.global_frame
print 'global relative frame', v.location.global_relative_frame

v.commands.goto(LocationGlobalRelative(v.location.global_frame.lat, v.location.global_frame.lon, 50))

print 'going to relative frame...'
time.sleep(10)

print 'global frame', v.location.global_frame
print 'global relative frame', v.location.global_relative_frame
```

It looks like it behaves normally: first going to +10 (takeoff), then +30 (via the global frame), then +50 (via the relative frame). Is there something wrong with my example?

hamishwillee, 2015-11-19:
> Is there something wrong with my example?

Yes. You've only given it 10 seconds to reach the global frame height, which is by co-incidence about 30m. If you change time after 'going to global frame...' to a minute you can watch it heading to 500+ metres, and then come back down to 50 metres :-)

tcr3dr, 2015-11-19:
New test:

```py
import time
from dronekit import connect, VehicleMode, LocationGlobal, LocationGlobalRelative

v = connect('tcp:127.0.0.1:5760', wait_ready=True)

v.parameters['FS_GCS_ENABLE'] = 0
v.parameters['FS_EKF_THRESH'] = 100

print 'await armable'
while not v.is_armable:
	time.sleep(.1)

print 'await guided'
v.mode = VehicleMode('GUIDED')
while v.mode != 'GUIDED':
	time.sleep(.1)

print 'await armed'
v.armed = True
while v.armed != True:
	time.sleep(.1)

print 'takeoff to 593'
v.commands.takeoff(10)

time.sleep(5)

print 'global frame:', v.location.global_frame
print 'global relative frame:', v.location.global_relative_frame
print ''

print 'global frame at 610...'
v.commands.goto(LocationGlobal(v.location.global_frame.lat, v.location.global_frame.lon, v.location.global_frame.alt + 20))
time.sleep(10)
print 'global frame:', v.location.global_frame
print 'global relative frame:', v.location.global_relative_frame
print ''

print 'relative frame at 630...'
v.commands.goto(LocationGlobalRelative(v.location.global_frame.lat, v.location.global_frame.lon, 50))
time.sleep(15)
print 'global frame:', v.location.global_frame
print 'global relative frame:', v.location.global_relative_frame
```

PR #469 (2015-11-19): Fixes vehicle.commands.goto with LocationGlobal.
Fixes #467.

In the way `.goto` is currently implemented, APM just ignores information about relative vs absolute location and defaults to absolute. This can be considered a bug, [or our use of MISSION_ITEM can be considered a hack](https://groups.google.com/forum/#!topic/drones-discuss/_UmqPM5b_60), and either way, we want a solution that works yesterday. For now, this restores the expected behavior, with minimal side effects (hopefully).
closed 2015-11-19 by tcr3dr