dronekit/dronekit-python #470: Future-proofing goto and takeoff

tcr3dr, 2015-11-19:
Last chance for future-proofing DKPY2!

We probably want the namespace `vehicle.commands.goto` and `.takeoff` in the future for waypoints. We may also want `vehicle.takeoff` and `vehicle.goto` as immediate commands with more complex behavior than what we support now.

I propose we change the current functions to `vehicle.simple_goto` and `vehicle.simple_takeoff` so that we

1. Preserve all examples that work now
2. Free up the namespace in the future to have more properly and thoughtfully implemented commands.

This is obviously weird to use the "simple" prefix but it would prevent us from doing a major version bump if we ever wanted to reclaim these function names. We’ll just deprecate (but not remove) the `simple_*` variants when we have suitable replacements.

hamishwillee, 2015-11-19:
@tcr3dr 

1. If we're not changing the API, why not just keep our old API "for now"? The namespace isn't taken and we don't end up with 2 deprecated APIs doing the same thing :-)

1. If we do add a simple_goto, I would still love this to have an additional argument ``speed`` which is set to None by default, and ignored if not set. This probably should bounds-check on max speed parameters (assuming there is one). If it is set then it should set the speed after setting the target.

  We should also create a speed attribute which gets the current speed and allows us to set it. There is a message for this I believe. It is still desirable to have it in the simple goto as I end up doing this most times.

1. I'd like to do the thinking for new API in this area ASAP. This is reason solo has to use so many custom messages. Clear shortfall iin the API.

hamishwillee, 2015-11-19:
@tcr3dr Just coming back here for others that might read this.

Yes, it does make sense to have simple_goto, but I still want a speed parameter and an attribute for speed. 

Re the namespace collision I now understand you wish to re-use commands.goto and commands.takeoff as simple mechanisms for adding the associated commands to the mission. Great idea! We should look at other common commands and see if there are good ways of adding them too - the current way is a bit of a pain because you really need to know the message formats ... and you shouldn't have to for the easy cases.

tcr3dr, 2015-11-20:
For better or worse: merged in #473!

closed 2015-11-20 by tcr3dr