dronekit/dronekit-python #475: Possible deadlock on connect()

atomictom, 2015-11-20:
I've found a strange bug. Calling `connect` hangs the program indefinitely when I use a udp connection string ("udp:172.28.128.1:5760", which is how I connect to a vagrant instance of SITL). I've traced it, and it appears to be caused by the `import array` on line 26 in pymavlink/generator/mavcrc.py. Moving the import to the top of the file seems to fix the issue. I suspect it may be a threading issue because that function is called on a separate thread. 

Also, I'm using the latest version of dronekit and pymavlink from pip, and this also occurs on dronekit master.

atomictom, 2015-11-20:
If I do `import imp` at the top, `imp.acquire_lock()` will hang the program.

atomictom, 2015-11-20:
This seems to be an issue with something I'm doing. I'll close this with the solution when I figure it out.

atomictom, 2015-11-20:
Soooo...I it turns out I was importing my main module (my entry point) from another module, [which is a big no-no](http://python-notes.curiousefficiency.org/en/latest/python_concepts/import_traps.html). I can only assume the double import was causing issues with the import machinery that only manifested itself in a non-top-level import (i.e. in a function) since all top-levels were probably done by then. Anyway, I'm closing this issue, sorry for the noise.

closed 2015-11-20 by atomictom

tcr3dr, 2015-11-23:
@atomictom No worries. Glad you tamed the Python import system, always a fun time.