dronekit/dronekit-python #484: Python3 support

conroy-cheers, 2015-11-27:
I find it rather odd that python3 seems to be unsupported, in a time when almost all other python modules support it.
Are there any plans for this?

hamishwillee, 2015-11-29:
No, though now that you have posted this we'll consider it for the roadmap. This is an open source project, so we'd also happily accept a push request which adds support for Python 3!

tcr3dr, 2015-11-30:
@conroy-cheers I agree with the sentiment, though most embedded targets where DKPY has historically been used have been Python 2-only.

Python 3 support would be lovely, so we'll leave this open as its tracking issue. We'd need:

1. Add `six` as a dependency or whatever features allow us 2.7 and 3.x support simultaneously
2. Update code to support 3.x
3. Update tests to run on Python 2 and Python 3 simultaneously

If this is something you have time to help with, as @hamishwillee mentioned, we will gladly work with you to make it happen.

Rosuav, 2016-03-11:
See the beginnings of support in PR #604. Please give it a try if you have Python 3 installed!

cclauss, 2016-10-04:
Python 3 builds are [currently breaking](https://travis-ci.org/dronekit/dronekit-python/jobs/164566497) because upstream [pymavlink](https://github.com/ArduPilot/pymavlink) has made a [relative import fix](https://github.com/ArduPilot/pymavlink/commit/918d936b03fdaf3140cb620db1b0e8bc796fec5d) that they have not yet pushed up to [PyPI](https://pypi.python.org/pypi/pymavlink/).

Evidlo, 2017-12-14:
Is Python 3 still unsupported?  There should probably be something about this in the README

cclauss, 2017-12-14:
Under Python 3 the syntax checking flake8 tests are passing but the functionality checking nose tests are not.  https://travis-ci.org/dronekit/dronekit-python

pietrodn, 2018-01-22:
I just submitted PR #789 together with a patch in pymavlink. With those edits, the nose integration and unit tests are passing.

pietrodn, 2018-04-01:
Update on this: DroneKit compatibility problems with Python 3 [have been solved](https://github.com/dronekit/dronekit-python/pull/789), but pymavlink still [needs patching](https://github.com/ArduPilot/pymavlink/pull/141).

pietrodn, 2018-10-31:
AFAIK, DroneKit fully supports Python 3; the unit and integration tests are automatically executed on each commit by CircleCI on Python 2 and Python 3. I think this issue can be closed.

peterbarker, 2018-10-31:
OK!

There's the one outstanding issue with parameters, which your patch does seek to solve.

closed 2018-10-31 by peterbarker

faxik, 2019-01-08:
The latest stable (2.9.1) still fails basic connect with errors:
>>> Exception in message handler for STATUSTEXT
>>> 'str' object has no attribute 'decode'
>>> Exception in message handler for STATUSTEXT
>>> 'str' object has no attribute 'decode'
>>> Exception in message handler for STATUSTEXT
>>> 'str' object has no attribute 'decode'

cclauss, 2019-01-08:
Do you know how to reliably reproduce this exception?  I might have messed up here:
* https://github.com/dronekit/dronekit-python/pull/705/files#diff-874cd63529f97ea08ecaa825961165ccR2842

pietrodn, 2019-01-08:
The latest stable published on PyPI (2.9.1) does _not_ support Python 3.
Try with the latest git master: it should be fully compatible.

faxik, 2019-01-08:
Understood, thank you!