dronekit/dronekit-python #494: "Integration tests" section in Contributing needs to be updated

atomictom, 2015-12-04:
A minor issue, but It looks like the example for the [integration test section](http://python.dronekit.io/contributing/contributions_api.html#integration-tests) is still based on DroneKit 1:

```python
def test_parameters(local_connect):
    v = local_connect().get_vehicles()[0]
```

hamishwillee, 2015-12-04:
Thanks. I'll fix that.

hamishwillee, 2015-12-04:
@tcr3dr I've updated  [the test doc](https://github.com/dronekit/dronekit-python/blob/hgw_test_update_dkpy2/docs/contributing/contributions_api.rst) in my branch - can you please sanity check this?

Also, actually running the test fails with ``No module named tests`` on Linux/Windows. Is there a different approach?

Lastly, what do you think about the idea of splitting out the tests either into their own single document or a folder with individual topics for the different test types. The contributing doc looks really cluttered.

hamishwillee, 2015-12-07:
This will be addressed in #495

mrpollo, 2015-12-07:
fixed by #495

closed 2015-12-07 by mrpollo