dronekit/dronekit-python #496: DK2 is_armable spotty

djnugent, 2015-12-07:
`is_armable` is spotty. Sometimes it says the vehicle is armable when it is not. 

I have been sending mode changes and then waiting for is_armable to be true using a while loop. Sometimes my while loop will exit before I am in an armable mode which then causes the copter to fail arming.
Example:
```
veh.mode = 'GUIDED'
while veh.is_armable  == False:
   time.sleep(0.5)
veh.arm = True
```

I have not looked into what is causing the issue

hamishwillee, 2015-12-08:
@djnugent Would be great if you could provide a reproducible test case (or do some debugging). Minimally I'd hope for a bit more information - are you using SITL or a real device? If you are using SITL, on what platform on what host? Does this happen often? Does it happen after initial startup or can it happen at any time...

djnugent, 2015-12-08:
It happens on sitl on a fresh start up. It fails like 5% percent of the time. It happened enough that it was noticeable. I will continue messing around with it to reproduce it more reliably.

djnugent, 2015-12-14:
I must have been doing something wrong. I haven't been able to reproduce. False alarm.

closed 2015-12-14 by djnugent

hamishwillee, 2015-12-14:
@djnugent I'm not sure you were, but I can't reliably reproduce either. Certainly this has not been a problem for me since I moved to a decent computer.