dronekit/dronekit-python #500: Basic pre-arm checks  not pass

idanbe, 2015-12-14:
code:
from dronekit import connect, VehicleMode
import time

vehicle=connect('/dev/ttyUSB0', baud=57600)

def arm_and_takeoff(aTargetAltitude):

    print "Basic pre-arm checks"
    # Don't let the user try to arm until autopilot is ready
    while not vehicle.is_armable:
        print " Waiting for vehicle to initialise..."
        time.sleep(1)

    print "Arming motors"
    # Copter should arm in GUIDED mode
    vehicle.mode    = VehicleMode("GUIDED")
    vehicle.armed   = True

    while not vehicle.armed:
        print " Waiting for arming..."
        time.sleep(1)

    print "Taking off!"

    # Wait until the vehicle reaches a safe height before processing the goto (otherwise the command
    #  after Vehicle.simple_takeoff will execute immediately).
    while True:
        print " Altitude: ", vehicle.location.global_relative_frame.alt
        #Break and return from function just below target altitude.
        if vehicle.location.global_relative_frame.alt>=aTargetAltitude*0.95:
            print "Reached target altitude"
            break
        time.sleep(1)

arm_and_takeoff(2)

terminal:
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
^CTraceback (most recent call last):
  File "takeOff.py", line 42, in <module>
    arm_and_takeoff(2)
  File "takeOff.py", line 19, in arm_and_takeoff
    time.sleep(1)
KeyboardInterrupt

hamishwillee, 2015-12-14:
This basically means what it says - your vehicle is not ready to arm. There are a lot of reasons that might be - you don't have a decent GPS signal would be the most common. 

You should check the mode (vehicle.mode) and gps_0 signal strength to get a better idea of what state your vehicle is in.

closed 2015-12-17 by idanbe

Biswajit1995, 2017-02-08:
Hello Hamishwillee,

First of all, I am a big fan of your's reply to several drone blogs. And I really appreciate that you spend your time to help people like us.

I am new to dronekit, and I want to use this for flying my drone indoors. For this I have attached PX4Flow camera with Teraranger One Lidar, and configured. But I am unable to arm in Guided mode as it shows need 3D-Fix.

--My questions:--
1. How to arm it in guided mode? It shows the same "Waiting for arming" message and not arming at all.
(Using the exact code given above)
2. I want to move my drone inside the building but have no idea how to give points as coordinates as I have no map and GPS wont work inside?

Note:  I know i am asking noobs questions, but i really need big help regarding this from you.

If you can mail me to this mail id : 
biswajit.sahoo4@tcs.com 
then it will be really a great help.

Thank you
Biswajit

hamishwillee, 2017-02-08:
Hi @Biswajit1995 

AFAIK you can't arm in guided mode, and in any case you MUST have GPS fix for the mode to work: http://ardupilot.org/copter/docs/arming_the_motors.html

ArduPilot doesn't really support "indoor flying" properly yet. The best place to discuss the options is gitter: https://gitter.im/dronekit/dronekit-python

Cheers
H

peterbarker, 2017-02-08:
On Wed, 8 Feb 2017, Biswajit1995 wrote:

>  2. I want to move my drone inside the building but have no idea how to give
>     points as coordinates as I have no map and GPS wont work inside?

Try GUIDED_NOGPS (you will need to be running master or one of the 3.5 
release candidates).  That means compiling your own arducopter binary and 
pointing dronekit-sitl at it.

hamishwillee, 2017-02-08:
Further to @peterbarker response, see http://ardupilot.org/copter/docs/ac2_guidedmode.html#guided-nogps

Biswajit1995, 2017-02-09:
Thank you for the reply guys

Ok so if i am unable to arm it in guided mode then is there any way i can accomplish my goal to "move my drone to given points inside building using dronekit python scripts on Raspberry pi"?

hamishwillee, 2017-02-09:
@Biswajit1995 Yes. So you put into the GUIDED_NOGPS mode - which you can do with DK_PYTHON. Then you can send movement messages that don't rely on GPS. Presumably the velocity commands that are relative to the device itself 
But as suggested previously, gitter is a much better place to ask for help.

Biswajit1995, 2017-02-09:
Thanks for the reply again,

As you told , I have asked I asked the question on Gitter.

But One problem I m facing is GUIDED_NOGPS will work on copter v3.4 and above but in copterv3.4 there is a bug for transmitter calibration, and no green bars are shown for calibration.

peterbarker, 2017-02-09:
On Wed, 8 Feb 2017, Biswajit1995 wrote:

> Thank you for the reply guys
> 
> Ok so if i am unable to arm it in guided mode then is there any way i can
> accomplish my goal to "move my drone to given points inside building using
> dronekit python scripts on Raspberry pi"?

You have some options.

Firstly, "given positions" implies you have an absolute position 
reference.  GPS gives you that.  ViCon gives you that.  SLAM gives you 
that.  Do you have that?

Optical flow only gives you relative velocities, not absolute position. 
You can dead reckon, but that can go hilariously wrong.

Biswajit1995, 2017-02-09:
> On Wed, 8 Feb 2017, Biswajit1995 wrote:
 Thank you for the reply guys

 Ok so if i am unable to arm it in guided mode then is there any way i can
 accomplish my goal to "move my drone to given points inside building using
 dronekit python scripts on Raspberry pi"?
You have some options.

Firstly, "given positions" implies you have an absolute position
reference.  GPS gives you that.  ViCon gives you that.  SLAM gives you
that.  Do you have that?

Optical flow only gives you relative velocities, not absolute position.
You can dead reckon, but that can go hilariously wrong.

Ok, So if i am using an Optical flow then 
1. Can I takeoff in Stabilize mode and move my drone to a certain distance like 1-2 meter forward then left or right ?
2. If, then what code should I refer?
Please, help.

peterbarker, 2017-02-09:
On Thu, 9 Feb 2017, Biswajit1995 wrote:

> Ok, So if i am using an Optical flow then
>
>  1. Can I takeoff in Stabilize mode and move my drone to a certain distance
>     like 1-2 meter forward then left or right ?

Yes, but this is the "relative velocity" case I was refering to; if your 
calibration changes your relative movements become "into wall" or " into 
potplants" rather than "down corridor.

You need an absolute position reference (from amongst the list I gave, or 
some other - e.g. a Pozyx system).

Peter

Biswajit1995, 2017-02-11:
Hey peter,

Thanks for the advise I will look into that. But I am facing another problem.

As I am using` optical flow camera` and a `lidar`,  I'm not able to arm my copter through this script.

I'm using this following code:

```
#/usr/bin/env python
# -*- coding: utf-8 -*-
from dronekit import connect,VehicleMode,LocationGlobalRelative
from pymavlink import mavutil
import time
# Receiving from command line
import argparse
parser = argparse.ArgumentParser()
parser.add_argument('--connect',help="PORT_NO")
args = parser.parse_args()

# Connecting to the vehicle
connection_string = args.connect
print("Connecting to...% s" % connection_string)
vehicle = connect(connection_string, wait_ready=True)

#Function to arm and takeoff to a specified altitude
def arm_and_takeoff(aTargetAlt):
	print("Basic Prearm checks..dont touch!!")
	print("Waiting for vehicle to initialize")
	time.sleep(1)
	print("Arming Motors..")
	# Copter should arm in Guided-mode
	vehicle.mode = VehicleMode("STABILIZE")
	
	while not vehicle.is_armable:
		print("Waiting for vehicle to initialize...")
		time.sleep(2)

	vehicle.armed = True
	vehicle.flush()
        print("Is Armed:% s"% vehicle.armed)
	
	while not vehicle.armed:
		print("Waiting for arming...Mode:% s"% vehicle.mode)
		print("Is Armed:% s"% vehicle.armed)
		vehicle.armed = True
		time.sleep(2)
		
	print("Taking Off..")
	vehicle.simple_takeOff(aTargetAlt)

	while True:
		print("Altitude: ",vehicle.location.global_relative_frame.alt)
		#Break and return from function just below target altitude.
		if vehicle.location.global_relative_frame.alt>=aTargetAlt: 
			print("Reached Target Altitude..")
			break
		time.sleep(2)
	
vehicle.close()
arm_and_takeoff(50)

```

But its printing:

```
Basic Prearm checks..dont touch!!
Waiting for vehicle to initialize
Arming motors..
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
```
And it keep printing like this, and not arming at all.
I don't know what I'm missing or doing wrong.
And to note as I have no gps connected, only PX4Flow and Lidar, that's why I'm trying to takeoff in STABILIZE mode.

Any help would be greatful..

Thanking you
Biswajit

erbaiaa, 2017-05-06:
- @Biswajit1995     I also encountered the same problem. Is your problem solved?

Biswajit1995, 2017-05-24:
Hello @erbaiaa ,

Yes the issue was solved as u have to remove this snippet of code from above code given by me
`while not vehicle.is_armable:
		print("Waiting for vehicle to initialize...")
		time.sleep(2)`
and try running again.  Hope this helps.

Thanks and Regards
Biswajit1995

Aryaelektro, 2017-08-04:
hello @Biswajit1995 I was tried your suggestion, but its printing:
`Connecting to vehicle on: /dev/ttyACM0
>>> Calibrating barometer
>>> ArduCopter V3.2.1 (36b405fb)
>>> Frame: QUAD
>>> barometer calibration complete
>>> GROUND START
Traceback (most recent call last):
  File "vehicle_state.py", line 39, in <module>
    vehicle.wait_ready('autopilot_version')
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2199, in wait_ready
    timeout)
dronekit.APIException: wait_ready experienced a timeout after 30 seconds.
`

Smark2019, 2022-07-07:
Hi,

I am also encountering this issue when I run this below snipped code:
`from dronekit import connect, VehicleMode, LocationGlobalRelative
from pymavlink import mavutil
import time

import argparse  
parser = argparse.ArgumentParser()
parser.add_argument('--connect', default='127.0.0.1:14550')
args = parser.parse_args()

# Connect to the Vehicle
print 'Connecting to vehicle on: %s' % args.connect
vehicle = connect(args.connect, baud=57600, wait_ready=True)

# Function to arm and then takeoff to a user specified altitude
def arm_and_takeoff(aTargetAltitude):

  print "Basic pre-arm checks"
  # Don't let the user try to arm until autopilot is ready
  while not vehicle.is_armable:
    print " Waiting for vehicle to initialise..."
    time.sleep(1)
        
  print "Arming motors"
  # Copter should arm in GUIDED mode
  vehicle.mode    = VehicleMode("GUIDED")
  vehicle.armed   = True

  while not vehicle.armed:
    print " Waiting for arming..."
    time.sleep(1)

  print "Taking off!"
  vehicle.simple_takeoff(aTargetAltitude) # Take off to target altitude

  # Check that vehicle has reached takeoff altitude
  while True:
    print " Altitude: ", vehicle.location.global_relative_frame.alt 
    #Break and return from function just below target altitude.        
    if vehicle.location.global_relative_frame.alt>=aTargetAltitude*0.95: 
      print "Reached target altitude"
      break
    time.sleep(1)

# Initialize the takeoff sequence to 20m
arm_and_takeoff(20)

print("Take off complete")

# Hover for 10 seconds
time.sleep(10)

print("Now let's land")
vehicle.mode = VehicleMode("LAND")

# Close vehicle object
vehicle.close()
`

Btw, I am running this script while executing px4 stil. Actually What I was expecting is that my drone would reach out 20 meter altitude and then landing. And I would be observing it all via Gazebo. But after running this above script, my quadcoper did not move anywhere. Here is my terminal output:

metin@metin-Lenovo-V330-15IKB:~/Desktop$ python takeoff_and_land.py 
Connecting to vehicle on: 127.0.0.1:14550
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
^CTraceback (most recent call last):
  File "takeoff_and_land.py", line 45, in <module>
    arm_and_takeoff(20)
  File "takeoff_and_land.py", line 21, in arm_and_takeoff
    time.sleep(1)
KeyboardInterrupt