dronekit/dronekit-python #501: GPS UTC timing

tnk3r, 2015-12-19:
How can I call the UTC time with dronekit? I can't seem to find any documentation about it.

hamishwillee, 2015-12-20:
@tnk3r What is it you are actually trying to do? There is no API to get current UTC time in dronekit itself, but plenty of APIs in Python.

tnk3r, 2015-12-20:
Right, I am looking at pymavlink. 

I am trying to sync multiple pixhawk together for timing and triggering actions. One obvious reason is syncing air and ground vehicle cameras with streaming cameras. They can easily be synced by providing initial timestamp when opening video stream. Then calculating offsets from one another.

hamishwillee, 2015-12-20:
@tnk3r We don't directly expose any timestamp information like that in DroneKit. However if there is a MAVLink message that is sent by the autopilot with this information you can intercept it in your dronekit app as described here: http://python.dronekit.io/examples/create_attribute.html

tnk3r, 2015-12-20:
Interesting, I'll give it a test. I'll report back soon.

hamishwillee, 2015-12-21:
Thanks. If you work out a good message and approach for synchronization we can add this to the API.

mikerob, 2015-12-21:
Having a simple function to set companion board date and time from GPS would also be useful - otherwise your times in logs are off.