dronekit/dronekit-python #502: Some Remarks and Questions on drone_delivery Example

deksprime, 2015-12-21:
Hello everyone, 

As I was getting quite deeper into Dronekit API, I would like to ask several questions about the specific example of Drone Delivery. 

Since at the time of writing Drone Delivery example was designed to work together with SITL (getting the heartbeat necessary to run the example after connecting to MAVProxy on SITL), the example works perfectly fine as intended when running it with SITL. To advance the example and realize it on a real vehicle, I have made a few modifications on the code where home location is changed, input of connect and launch methods are modified. I have also added a Return To Launch page where the simulated vehicle actually returns to launch when invoked. After trying and succeeding on simulated version of the example, I have tried to connect to an APM flight controller through the Drone Delivery example, though I had to run MAVProxy and connect to the vehicle and get the necessary parameters, otherwise the example won’t work (motors won’t arm, copter won’t pass the pre-arm check). I could manage to successfully take off using Drone Delivery example after I’ve got the parameters from the MAVProxy and change the flight mode to “GUIDED” by writing ‘mode guided’ on the MAVProxy command line, and shutting down the MAVProxy before running the example. I have a few questions on these remarks: 

What could be the possible reason for the example’s not working when I do not run MAVProxy before running the example?

Given the fact that I could manage to get the drone to arm and take off from the example, what could be the possible reason for the “Go” not working while the quad hovers? (it is useful to mention that I’m able to make it work on the example running with SITL, and I also think that my UAV has no problems related to connection, getting MAVLink messages etc. since I am able to use all the features of MAVProxy on the UAV.)

Since there was no “Land” or “RTL” buttons on the original example, I would like to ask whether if this example was intended to work on a real UAV, or was it designed to work solely together with SITL. If so, your advice and suggestions on what to add and what to focus on would really help developing the example and enabling it for a real UAV usage.
[drone_delivery.py.zip](https://github.com/dronekit/dronekit-python/files/69022/drone_delivery.py.zip)
[command.html.zip](https://github.com/dronekit/dronekit-python/files/69024/command.html.zip)
[landit.html.zip](https://github.com/dronekit/dronekit-python/files/69025/landit.html.zip)

hamishwillee, 2015-12-21:
Hi @deksprime 

Firstly, thank you for taking some time with the example. As you will note, there is a [warning on the documentation](http://python.dronekit.io/examples/drone_delivery.html) of problems with this one. While the associated defect has been closed, I have not retested since that was posted.

The example was written to demonstrate integration with a web server. It is an example - not a fully fledged "product". That said, I think minimally it needs to be updated to at least get its initial location from the home location (as suggested in #362). I think it would be good to add an RTL/LAND option as you have done too.

I do not "know" why you'd be having problems. First modify this code
```
# Connect to the Vehicle
print 'Connecting to vehicle on: %s' % args.connect
vehicle = connect(args.connect, baud=57600)
```
to 
```
# Connect to the Vehicle
print 'Connecting to vehicle on: %s' % args.connect
vehicle = connect(args.connect, wait_ready=True, baud=57600)
```

You have also removed this line which is important since it tells you that the vehicle is ready to arm. If you don't have the line then you may be sending the arm command when the vehcle is not ready - and the command will be silently dropped.
```
while not self.vehicle.is_armable:
            time.sleep(.1)
```
If you want to find out *why* it isn't arming, then do something like:

```
if v.mode.name == "INITIALISING":
    print "Waiting for vehicle to initialise"
    time.sleep(1)
while vehicle.gps_0.fix_type < 2:
    print "Waiting for GPS...:", vehicle.gps_0.fix_type
    time.sleep(1)
while not self.vehicle.is_armable:
            time.sleep(.1)
```

Let me know how that goes!

PGSK1234, 2021-07-18:
I see this on http://localhost:8080/track and http://localhost:8080/command. Any idea why I dont see the map? Thanks.
[image]
[image]