dronekit/dronekit-python #503: Add AUX output controls

a7scenario, 2015-12-22:
For servos on the Aux outputs, it would be great if DroneKit had functionality for controlling Aux servos similar to how RC Overrides are handled.  The only way, as far as I know, to control servos/sensors on the Aux Outputs is to use DK's message factory to make a custom MAV_CMD/DO_SET_SERVO mavlink message.  
For example, syntax like vehicle.channels.servo['11']="2000" would be much simpler than using the message factory.

hamishwillee, 2015-12-22:
Hi @a7scenario that certainly makes sense to me (also equivalent in relay). Need to consider also the servo_relay repeat functions and what they mean in terms of the Python API. 

I doubt that we will get around to this quickly, but if you have time, you might want  to create a patch for this that we would review and integrate.

tucher, 2016-01-17:
Yes, please, implement that! It would be very useful.

lavvy, 2018-04-10:
Any update on this