dronekit/dronekit-python #506: Add support for Emergency Stop

a7scenario, 2015-12-23:
It would be useful if DroneKit has built-in support for the Motor Emergency Stop functionality introduced in 3.3 Firmware.  

https://www.youtube.com/watch?v=-Db4u8LJE5w

hamishwillee, 2015-12-28:
This can be implemented as a COMMAND_LONG with a MAV_CMD_DO_FLIGHTTERMINATION message from Copter 3.3

@a7scenario When do you think you'd call this programmatically (under what circumstances?)

@tcr3dr We kind of need a strategy for handling vehicle-version specific functionality. Is that another class/set of classes? So VehicleCopter3.2, VehicleCopter3.3 etc?

a7scenario, 2015-12-29:
I would imagine would may have safety-related applications.  For the time being, I'll use the MAV_CMD_DO_FLIGHTTERMINATION msg -- thanks for pointing that out.  

For me, I'm looking to try it for precision landing as a means to immediately shut-off the motors when directly overhead the landing target at very low altitude, instead of trying to change the loiter target at low altitude and introducing the jerk associated with that.

hamishwillee, 2015-12-29:
Hi @a7scenario 
I would discuss the options for precision landing on the gitter channel - up to you, but just terminating the flight at the last minute may not be the best approach (I simply don't know). I do know that there is work going on to support precision landing, including new messages etc planned for Copter 3.4 (maybe)

As for previously, this is unlikely to happen "immediately" but very happy to take patches and we will review.