dronekit/dronekit-python #511: [Feature] Obtaining Flight Log via Dronekit-python

sahiljuneja, 2016-01-09:
Hello,

When DKP was reliant on MAVProxy, the flight logs were saved to a local folder from where we were trying to run MAVProxy + DKP code. I was wondering if something similar is or could be easily set up with DKP2 or not?

If it's possible to do it, but not under recent development or with that intention, perhaps I could give it a shot of implementing it if pointed in the right direction. 

Please let me know what you think works best, if at all.

hamishwillee, 2016-01-19:
@sahiljuneja I think this could be done by sending  [custom MAVLink log messages](http://mavlink.org/messages/common#LOG_REQUEST_LIST) to request the logs, and `onmessage()` to intercept the responses.

But (personally) don't think it is something that DK-Python should support. If you need logs you can use a GCS to get them. 

The reason for this opinion is that DK-Python is mainly intended for driving the autopilot from a companion computer - unless you're planning on doing on-the-fly analysis of logs there really isn't a need for logs via dk. 

I don't see us implementing the functionality. If you do see a strong use case or have code to contribute we would certainly consider it though. Thanks!

shubham9436, 2021-02-25:
is there a way to download all the logs using dronekit-python?