dronekit/dronekit-python #512: Vehicle.battery returns 0.0 voltage and "None" current repeatedly

sahiljuneja, 2016-01-11:
Hello,

I am trying to utilize vehicle.battery command to obtain the battery status from my drone. However even over a period of time I get zero voltage.  

Is there any particular reason for this to happen?

sahiljuneja, 2016-01-12:
Didn't know I have to enable Battery Monitoring via Mission Planner for this to work.

closed 2016-01-12 by sahiljuneja

hamishwillee, 2016-01-19:
@sahiljuneja Exactly what did you need to do? DroneKit simply converts the battery status messages it receives from the vehicle into an attribute - so provided the messages are sent you will get values.

sahiljuneja, 2016-01-19:
@hamishwillee That's exactly what I was trying to do. But as I mentioned above, I had to enable "Battery Monitoring" option from the Mission Planner so that the drone can actually send back battery information.  It's working properly after I enabled that.