dronekit/dronekit-python #514: Attempts to change home_location, not working

sahiljuneja, 2016-01-14:
Hello,

As per DKP2.0, we have the option to modify the home_location.

I have tried changing the home_location, but nothing happens. I have tried 

> vehicle.home_location.lat = new_location.lat
> vehicle.home_location.lon = new_location.lon

Or,

> new_location = LocationGlobal(lat_value, lon_value) 
> vehicle.home_location = new_location

Left alt as None here, also tried to use the same alt as the initial home_location. Didn't work. However, the above threw an error because I couldn't use float values for lat_value and lon_value. Why is that though? I came across the above in a script which was meant for SITL. Is that the reason? 

Based on "Setting Attributes" on http://python.dronekit.io/guide/vehicle_state_and_parameters.html

> Commands to change a value are not guaranteed to succeed (or even to be received) and code should be written with this in mind.

So am I required to wait for the pixhawk to accept the change? Whatever I tried from the above, didn't really post an error. When I tried to read off the new home_location after setting it to a new one, it just returned me the original home_location. So what's required for this? (Seems an important issue with this added feature as per me)

hamishwillee, 2016-01-19:
Does the home location change when you run the example code [supplied with dronekit here](http://python.dronekit.io/examples/vehicle_state.html). As you can see, this just gets the current location and changes the alt value so that we can observe that it has indeed changed.

```
my_location_alt=vehicle.location.global_frame
my_location_alt.alt=222.0
vehicle.home_location=my_location_alt
```

I would expect your second approach to work:
```
new_location = LocationGlobal(lat_value, lon_value) 
vehicle.home_location = new_location
```
I don't think just changing like this would work, but I'd be interested in confirmation:
```
vehicle.home_location.lat = new_location.lat
```

If the example code works, and yours does not, then the most likely cause is that the home location cannot be set until AFTER it has been set by the autopilot initially (after acquiring GPS lock). Before this, any changes you make will be ignored.

sahiljuneja, 2016-01-19:
> I would expect your second approach to work:

That raised errors because LocationGlobal is not accepting float values.

> then the most likely cause is that the home location cannot be set until AFTER it has been set by the autopilot initially

I used the code to check for this as mentioned in the vehicle_state example. It waits to get a GPS lock, then returns the home position, I try to change the home_location by the different methods I mention above and nothing happens.

I will try changing just the altitude as you point out in that example. But I don't know why none of the methods above don't work at all.

hamishwillee, 2016-01-20:
```
vehicle.home_location.lat = new_location.lat
```
This does not work because the home_location is the attribute, and it changes on that value which are detected and cause the new home location to be uploaded (not changes to the members of the attribute). I think it would be "nice" if you could change the home location like this, but you can't currently.

The other code isn't working because you are missing the altitude. This is wrongly being reported as a float being not supported:
```
new_location = LocationGlobal(lat_value, lon_value, #you need altitude too - there is no default!) 
```
It is also possible for ANY command to silently fail - either by not being received, or because it is invalid and the autopilot chooses to ignore it.

hamishwillee, 2016-02-19:
This has been resolved IMO by my response.

closed 2016-02-19 by hamishwillee

gusisin, 2018-06-14:
I am having problems setting home. I calculate a new location, I set as in the example above but the new location does not stick as you can see from the output. I copied the set home code only changing the new location calculation, much head scratching.  here is the relevant code
`    if mode == 'TUBED':
        print('Just sitting here enjoying life ')
        #--- Wait until the button is pushed.
        time.sleep(2)
        # Get Vehicle Home location - will be `None` until first set by autopilot
        while not vehicle.home_location:
            print('No vehicle location ')
            cmds = vehicle.commands
            cmds.download()
            cmds.wait_ready()
            print(vehicle.home_location)
            if not vehicle.home_location:
                print('Waiting for home location ...')

        # Home location must be within 50km of EKF home location (or setting will fail silently)
        # In this case, just set value to current location with an easily recognisable altitude (222)
        my_location_alt = vehicle.location.global_frame
        my_location_alt.alt = 222.0

        print("Old Loc %s" % vehicle.home_location)
        
        #calculate new location
        newloc = get_location_metres(my_location_alt, 40, 40)
        print("new Loc %s" % newloc)
        
        #update the location
        vehicle.home_location = newloc 
        
        #Confirm current value on vehicle by re-downloading commands
        cmds = vehicle.commands
        cmds.download()
        cmds.wait_ready()

        if check_button():
            print ("Prepare a flight plan based on what we know")
            print(home.Lat)
            mode = 'PREP'`
The output is 
`mob@mobdev:~/scripts$ sudo python life.py
Connecting...
>>> APM:Copter V3.3 (d6053245)
>>> Frame: QUAD
Just sitting in the tube enjoying life 
No vehicle location 
LocationGlobal:lat=51.637928009,lon=-0.471892386675,alt=0.0
Old Loc LocationGlobal:lat=51.637928009,lon=-0.471892386675,alt=0.0
new Loc LocationGlobal:lat=51.6379288261,lon=-0.471892433577,alt=0.0
Just sitting in the tube enjoying life 
Old Loc LocationGlobal:lat=51.637928009,lon=-0.471892386675,alt=0.0
new Loc LocationGlobal:lat=51.6379288261,lon=-0.471892433577,alt=0.0
Just sitting in the tube enjoying life 
Old Loc LocationGlobal:lat=51.637928009,lon=-0.471892386675,alt=0.0
new Loc LocationGlobal:lat=51.6379288261,lon=-0.471892433577,alt=0.0
Just sitting in the tube enjoying life`