dronekit/dronekit-python #516: Use a single communication link for several vehicles

jlecoeur, 2016-01-22:
It seems DroneKit promotes one link per vehicle and does not work well when more than one vehicle send messages on the same link.

What is the best way to proceed?

jlecoeur, 2016-01-22:
In my setup, several vehicles communicate on the same network. As a test, I started two simulated drones (respectively sysid 1 and 2) on my computer, both sending messages to udp:127.0.0.1:14550. Then I created a vehicle object in dronekit:
```
import dronekit
vehicle1 = dronekit.connect('127.0.0.1:14550')
```
All the Vehicle methods decorated by the `on_message` decorator are called for messages coming from drone with sysid 1 **AND** the drone with sysid 2. 
For instance `vehicle1._heartbeat_system` keeps on oscillating between 1 and 2. 
Similarly, this leads to undefined behaviour when both drones send 'PARAM_VALUE' messages.

jlecoeur, 2016-01-22:
I partially solved the problem by:
- specifying the target_system at instantiation of a Vehicle object, 
- in MAVconnection, check that 
```(self.target_system == 0) or (self.target_system == msg.get_srcSystem()``` before calling the message listener functions

I could create a vehicle like this 
```
import dronekit
vehicle1 = dronekit.connect('127.0.0.1:14550', target_system=1)
```
And it only received messages from the drone with sysid 1, which is what we want, great!

However if then I create a second vehicle object:
```
vehicle2 = dronekit.connect('127.0.0.1:14550', target_system=2)
```
Then the link to the first drone is lost. This comes down to `mavutil.mavlink_connection`s who cannot share the same udp link.

Maybe the `connect` function should create a single MAVConnection object, but return a list of vehicles? I am happy to contribute to that feature, but I would like to be sure I go in the right direction.

hamishwillee, 2016-01-25:
Hi @jlecoeur 

What you're doing here is not talking to two separate drones -  but taking to one drone twice from separate dronekit instances. This is possible but largely pointless - to make it work you need to keep the the target system the same in both cases (since this is the ID of the drone on the mavlink network) and you "might" change your source id.

What I think you actually want to do is talk to two separate drones. To do this you need to attached to a real or simulated drone on a different port. I believe this can be done by passing a value to sim_vehicle.sh when you launch SITL. The help options to that script should help you work out which.

jlecoeur, 2016-01-25:
Hi,

Indeed I want to talk to several drones.
However, when talking to real drones, they will be using the same xbee network, so messages from N drones will be coming through the same link.
With current implementation, I would have to write a MAVLINK demultiplexed that reads on xbee, and forwards messages to N udp ports according to the sysid of the incoming message. 
It would be a lot more convenient if dronekit could directly feed messages to a list of vehicle instances according to the sysid.

mrpollo, 2016-01-26:
hey @jlecoeur can you please write a test case that proves your issue?

jlecoeur, 2016-01-26:
Sure! Here is a testcase using 2 simulated mavric autopilots: 
https://gist.github.com/jlecoeur/ab85e970b3b048985273
The gist contains terminal outputs, but you should be able to replicate it by running the script testcase.sh. The example requires 3 terminals:
- TERM0: runs dronekit script
- TERM1: runs MAV with sysid 1
- TERM2: runs MAV with sysid 2

What it does:
- [TERM0] Compile MAV code, start python script, wait user input
- [TERM1] Start emulated MAV with sysid 1
- [TERM0] Get parameter value -> receive 0.0
- [TERM0] Set parameter to 1.0
- [TERM1] Notifies the parameter is set to 1.0
- [TERM0] Get parameter value -> receive 1.0
- [TERM2] Start emulated MAV with sysid 2
- [TERM0] Get parameter value -> receive 0.0 **(ERROR)**
- [TERM0] Set parameter to 1.0
- [TERM1] Notifies the parameter is set to 1.0
- [TERM0] Get parameter value -> receives 1.0

After MAV2 is started, dronekit accepts the 'PARAM_VALUE' message from MAV2, and the parameter is set to 0.0 in the dronekit object, although it is still equal to 1.0 onboard MAV1.

jlecoeur, 2016-02-17:
@hamishwillee @mrpollo any update on this?

hamishwillee, 2016-02-18:
@jlecoeur Not as far as I know. Working on other priorities. @peterbarker , do have time in the coming weeks to look at this?

rmackay9, 2016-02-22:
I agree this is an issue.  We should be able to speak to multiple vehicles using a single connection.  It's normal for an antenna tracker and vehicle to appear to be on the same connection from dronekit's point of view.  The vehicles can be distinguished by their sysid/component-id combination.

hamishwillee, 2016-02-22:
My two bits 
* we need to filter the incoming messages for a vehicle so that only those with the  ``source_system`` (or broadcast IDs) set in the [connect()](http://python.dronekit.io/automodule.html#dronekit.connect) are processed by the vehicle. This makes sense because  ``source_system`` is the system ID of the vehicle and it should only be working on messages intended for it. That should sort out the test case above.
* We also need to look at allowing the target_id to be set in the connect method so that we can choose what the default target is for messages from this vehicle. At the moment I *think* all messages are sent to 0 (the broadcast ID).

rmackay9, 2016-02-23:
I hit this problem again today in a more common situation.  I'm using dronekit on a companion computer and the vehicle.mode is switching between the vehicle's actual mode (Guided, Loiter, etc) to Stabilize.  the reason is that I have Mission Planner attached as well and it's sending heartbeats which include the flight mode.  Of course mission planner's flight mode is not what I want! :-)

hamishwillee, 2016-02-23:
Very annoying. At least for things like mode setting we really need to check that the source is our connected vehicle. @peterbarker do you want me to take this issue over. I can look probably fix it early next week.

peterbarker, 2016-03-15:
Rough instructions for multiivehicle support in dronekit-python: https://github.com/peterbarker/dronekit-python/tree/source-system-filtering/examples/multivehicle

jainakshay91, 2016-04-17:
So, this is a side note, but @jlecoeur if you are using the xbee network, then I think you can filter your messages for separate vehicles based on their MAC address/IP address depending on the type of xbees you are using. This should help you to alleviate any issues with the dronekit-python library and allow you to communicate with multiple vehicles at one time.

basil-huber, 2016-10-17:
Hi, are there any plans to merge @peterbarker 's solution into master?

tlillis, 2017-07-06:
Hello, has there been any progress on this? Any plans for the merge of @peterbarker 's stuff? Thanks!

ajaiRevision, 2018-10-08:
@peterbarker ,
Hi Peter,
I am having trouble getting your example case given in https://github.com/peterbarker/dronekit-python/tree/source-system-filtering/examples/multivehicle to work. 

I had ran all the commands until the last command in step 5:
python multivehicle.py  --simulation-count 3 --extra-connection udpout:localhost:9432

This gave an error below. Could you provide some insight on this?

[image]

peterbarker, 2018-10-17:
On Mon, 8 Oct 2018, ajaiRevision wrote:

> I had ran all the commands until the last command in step 5:
> python multivehicle.py --simulation-count 3 --extra-connection udpout:localhost:9432
> 
> This gave an error below. Could you provide some insight on this?

Wrong pymavlink; try setting PYTHONPATH to point to the relevant git 
repository, checked out on the correct branch.

Peter

maroquio, 2018-12-21:
It's very disappointing dronekit not having a multi-vehicle protocol. There is a lot of research that uses microdrone swarms that could use this feature. I hope you have time to add this feature soon.

lauracanseco21, 2019-06-26:
@peterbarker 
I tried running your source system filtering, but it doesn't differentiate the two drones i am working with. how should i go about fixing this issue?

mbatra9, 2019-07-02:
Hi, I have a very minor issue that hopefully someone will be able to solve. I'm following the instructions on the ReadMe page for the github site Peter Barker posted with his example code. On step 1, I typed this line into my terminal "git checkout peterbarker/source-system-filtering" and I received an error message saying that I don't have access to this repository. Anyone know what I'm doing wrong? 
Am really trying to figure out how to get multivehicles set up in dronekit so any help would be appreciated.

bilginmu, 2020-11-09:
I tried the @peterbarker repo but I got errors. Is there any way to use a single commnication link for several vehicles ?

GRS26, 2022-05-30:
Also having problems while following the instructions, idk if already having dronekit and pymavlink installed though pip is affecting the process

commit 94f5b62a (2016-02-24): Support filtering incoming messages based on target_id

Closes #516
commit f556c5ae (2016-10-18): Support filtering incoming messages based on target_id

Closes #516