dronekit/dronekit-python #524: No module named dronekit - 3DR Solo - Running code onboard - Help needed

reyes73a, 2016-02-02:
Hi guys,
I am having a little trouble running the helloworld example for solo developers (http://dev.3dr.com/example-helloworld.html) and I was hoping some one could shed some light on this for me. I have been following the development guide to run solo scripts onboard. I can run helloworld.py from my PC using:

```
git clone https://github.com/3drobotics/solodevguide1
cd solodevguide/examples/helloworld
python helloworld.py
```

This runs fine without any issues, but when I try to run it onboard solo it fails. I do the following commands:

`solo script pack`

This command successfully creates the file: solo-script.tar.gz
I then try to run:

`solo script run helloworld.py`

but it fails with the following error:

```
' File "/log/solo-script/helloworld.py", line 1, in from dronekit import connect
ImportError: No module named dronekit
```
Since I can run it on the ground station pc without any issues I am guessing that the solo script run command is failing because I am missing some files on my solo? I thought dronekit was already included onboard... I am currently lost on how to fix this so any suggestions would be greatly appreciated! Thanks in advance!

hamishwillee, 2016-02-02:
Hi @reyes73a 

First of all, it doesn't sound to me like you're doing anything wrong (I wrote the instructions, and tested them at that point). The pack command should include dronekit in the package as a requirement. The examples should work without any issue. 

Can you 
1. Confirm what host operating system you are using
2. Put the full log for running `solo script run helloworld.py`  (or confirm that all that is displayed is that error)

Regards
H

PS Just for future reference, best to post Solo specific questions in the [solodevguide repo](https://github.com/3drobotics/solodevguide/issues). This is almost certainly a packaging problem, and not a dronekit problem as such.

hamishwillee, 2016-02-03:
Can you also please attach your generated solo-script.tar.gz and confirm that the directory you are packaging contains the *requirements.txt* file.

reyes73a, 2016-02-04:
Thanks a lot for your reply Hamish!

1. I am running Ubuntu 14.04.3 LTS (trusty)
2. Full error log is attached. Trying to attach solo-script.tar.gz but github is not allowing me to. Also tried attaching it as ZIP but no luck. I can see the contents of solo-script.tar.gz:
**solodevguide   
.gitignore   
dronekit-2.0.2-py2-none-any.whl   
helloworld.py   
pymavlink-1.1.69-py2-none-any.whl  
README.md   
requests-2.9.1-py2.py3-none-any.whl
requirements.txt**

**requirements.txt** contains "dronekit>=2.0.0")

Also the directory structure where the package and run commands are being executed is:
**/solodevguide/examples/helloworld**

Inside /helloworld there is the following:
**helloworld.py   README.md   requirements.txt   solodevguide   solo-script.tar.gz**

If I open requirements.txt it also contains only one line of text:
**dronekit>=2.0.0**

[errorlog.txt](https://github.com/dronekit/dronekit-python/files/117792/errorlog.txt)

Any other way I can send you the solo-script.tar.gz?

Thanks a lot for your help. I really appreciate it.

hamishwillee, 2016-02-05:
@reyes73a Can you please email me the tarball - hamish at 3drobotics dot com .

reyes73a, 2016-02-05:
Just emailed you the tarball. Thanks!

reyes73a, 2016-02-05:
According to what I read above, I see that peterbarker committed a fix for this? What is the proper way to take in this fix? Thanks!

hamishwillee, 2016-02-08:
FYI, I replied to @reyes73a offline. When the icon at top of page says "Merged" rather than "Open" they can following the [update instructions](http://dev.3dr.com/starting-utils.html) (use -U flag to pip) to get the latest version.

hamishwillee, 2016-02-11:
@reyes73a FYI, we still haven't managed to test the fix (https://github.com/3drobotics/solo-cli/pull/37) on Mac OS X, so haven't been able to make a release. I'm updating a mac to a new enough version (hopefully) for testing.

reyes73a, 2016-02-11:
> test the fix (3drobotics/solo-cli#37

Yes! I noticed on git hub. Thats ok. I will remain on hold until the merge
happens. Thanks for the update!

nicOwlas, 2016-02-16:
Hi @reyes73a and @hamishwillee ,

Glad that I found you; I have exactly the same issue on Ubuntu, nevertheless it works from cygwin... Still there are errors (see code below) output, but it runs. I am looking forward the merge status.

Nevertheless the command `python helloworld.py` always fails, it says `>>> Link timeout, no heartbeat in last 5 seconds`.

@hamishwillee do you have any idea why? Strangely enough Mission Planner always succeed connecting to the solo and reading the parameters. It is as if the connect functions between MP and dronekit were different.

On cygwin:
`$ solo script run helloworld.py
connecting to solo...
checking pip...
uploading script bundle...
./
./.gitignore
./helloworld.py
./README.md
./requirements.txt
./wheelhouse/
./wheelhouse/dronekit-2.0.2-py2-none-any.whl
./wheelhouse/pymavlink-1.1.70-py2-none-any.whl
./wheelhouse/requests-2.9.1-py2.py3-none-any.whl
Not deleting env/bin
New python executable in env/bin/python
Installing setuptools, pip, wheel...done.
cp: can't stat '/usr/lib/python2.7/site-packages/greenlet*': No such file or directory
Ignoring indexes: https://pypi.python.org/simple
Processing ./wheelhouse/dronekit-2.0.2-py2-none-any.whl
Processing ./wheelhouse/pymavlink-1.1.70-py2-none-any.whl
Processing ./wheelhouse/requests-2.9.1-py2.py3-none-any.whl
Installing collected packages: pymavlink, requests, dronekit
Successfully installed dronekit-2.0.2 pymavlink-1.1.70 requests-2.9.1
running script...

ERROR LOADING MAVNATIVE - falling back to python implementation
Connecting to udpin:0.0.0.0:14550...
Vehicle state:
 Global Location: LocationGlobal:lat=47.2819165,lon=-1.5205414,alt=2.98
 Global Location (relative altitude): LocationGlobalRelative:lat=47.2819165,lon=-1.5205414,alt=2.98
 Local Location: LocationLocal:north=None,east=None,down=None
 Attitude: Attitude:pitch=0.00435095559806,yaw=1.80949473381,roll=-0.00715055828914
 Velocity: [0.059999999999999998, 0.02, 0.0]
 Battery: Battery:voltage=15.607,current=0.91,level=66
 Last Heartbeat: 0.0524210929871
 Heading: 103
 Groundspeed: 0.399999976158
 Airspeed: 0.399999976158
 Mode: LOITER
 Is Armable?: False
 Armed: False
Done.
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", line 84, in mavlink_thread
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 1092, in listener
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'time'
`
Thanks in advance,

hamishwillee, 2016-02-16:
@nicOwlas I am confused - your log above essentially shows success - there is an exception, but it is after the thread completes. I certainly can't see ``>>> Link timeout, no heartbeat in last 5 seconds.``

I think you're saying that if you run ``python helloworld.py`` on the ground station you get a timeout? Did you first connect to solo and use its IP address in the above command?

nicOwlas, 2016-02-16:
Hi @hamishwillee,

sorry my mail was confusing. The thread corresponds to running the script on-board the solo and executes correctly (exception for the `ERROR LOADING MAVNATIVE`?).

The main problem I am facing is that I cannot start a script launched from the ground station, the error is then always the same: `>>> Link timeout, no heartbeat in last 5 seconds.`.
I have tried various launch command (from ubuntu or cygwin):
`python helloworld.py`
`python2.7 helloworld.py`
`python helloworld.py udpin:0.0.0.0:14550` (also trying with other ports up to 14560)

It fails consistently. As I was saying, strangely mission planner manages to connect (UDP on port 14550).

hamishwillee, 2016-02-17:
@nicOwlas 

>  (exception for the ERROR LOADING MAVNATIVE?).

Yes, I get this too. We should kill that warning, but it can be ignored for now.

When connecting from the PC, either of these should work provided your PC is connected to Solo's wifi network. That said, I have not tried form Cygwin, only from normal Python prompt.
```
`python helloworld.py`
`python helloworld.py udpin:0.0.0.0:14550
```

I will try to test this in the coming days ...

Regards
Hamish

hamishwillee, 2016-02-17:
@reyes73a We've updated the version of solo-cli on master, so you should be able to test this now. There are still some odd artifacts when helloworld.py is being unpacked and after it runs, but these can be ignored - it works! We're now looking at adding full windows support for the pack operation.

reyes73a, 2016-02-17:
Thanks Hamish! Let me give it a try and get back to you! Thanks again!

nicOwlas, 2016-02-18:
Hi @hamishwillee,

Thanks a lot for the reply and the update of the solo-cli. I could test it on windows (cygwin) and Ubuntu. The results:
- cygwin: everything works, onboard and offboard :-)
- ubuntu: it works onboard only. I still get the same error while executing `python helloworld.py` from the terminal or from spyder...
'>>> Link timeout, no heartbeat in last 5 seconds'
'>>> No heartbeat in 30 seconds, aborting.'

This is pretty strange as @reyes73a for you, on Ubuntu you had no issue with running a script offboard... Where do you write the `python helloworld.py` command? In the terminal directly or by a run in Spyder? Do you ssh on solo beforehand? Do you have the solo wifi command enabled?

Cheers,

reyes73a, 2016-02-18:
Hi @hamishwillee  and @nicOwlas ,

I updated solo-cli and the **script run** command now runs correctly. However once the script runs I get timeout issues:

connecting to solo...
checking pip...
script bundle already up to date.
running script...
>>> Link timeout, no heartbeat in last 5 seconds
ERROR LOADING MAVNATIVE - falling back to python implementation
Connecting to udpin:0.0.0.0:14550...
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "/log/solo-script/helloworld.py", line 7, in <module>
    vehicle = connect(target, wait_ready=True)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2769, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2048, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.

I have tried running this with solo hooked up to the internet and also without internet and I still get the same result. **I can still run the helloworld.py offboard without issues in two different ways:**
**python helloworld.py** and **python helloworld.py udpin:0.0.0.0:14550**

@nicOwlas , when running the script offboard through Ubuntu I simply connect to solo's wifi (I can have solo connected to the internet or not - it does not affect the end result for me) then:

**cd devsologuide/examples/helloworld**
**python helloworld.py** or **python helloworld.py udpin 0.0.0.0:14550**

I use the regular linux terminal and I do not SSH into solo at all.

@hamishwillee  Any thoughts on this timeout issues? Very strange indeed... Thanks =)

hamishwillee, 2016-02-19:
@reyes73a No idea. Can you please:
1. Confirm what version of Solo you are using - attach output of `solo info`. 
2. Email me with the tarball that was created when you did pack (hamish at 3drobotics dot com)
3. Modify the example in the first line from `vehicle = connect(target, wait_ready=True)` to `vehicle = connect(target, wait_ready=False)`. I'd like to see if the failure is in connecting or getting the parameters.

Thanks

hamishwillee, 2016-02-19:
@reyes73a Also, can you please update Solo to the latest version. It is mostly likely that you've got a version that doesn't redirect to the UDP port internally.

reyes73a, 2016-02-22:
Hamish,

1. When I run the Solo App I see under settings the following info:

- Solo 1.3.0

- Autopilot 1.2.19

- Gimbal 1.2.0

- Controller 1.3.0

- Controller Firmware 1.2.6

The App says I am up to date.

When I run solo info from linux I get:
```
connecting to Solo and the Controller...
{
  "controller": {
    "ref": "3dr-controller-imx6solo_3dr_artoo-20151118202515", 
    "version": "1.3.0"
  }, 
  "gimbal": {
    "connected": true, 
    "version": "1.2.0"
  }, 
  "pixhawk": {
    "apm_ref": "331103b2", 
    "px4firmware_ref": "60133536", 
    "px4nuttx_ref": "d48fa307", 
    "version": "1.2.19"
  }, 
  "solo": {
    "ref": "3dr-solo-imx6solo_3dr_1080p-20151118202712", 
    "version": "1.3.0"
  }
}
```
2 Just emailed you the new tar file.
3 After modifying the example I solo pack and then solo run again. I get the same results:

```
running script...

>>> Link timeout, no heartbeat in last 5 seconds
ERROR LOADING MAVNATIVE - falling back to python implementation
Connecting to udpin:0.0.0.0:14550...
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "/log/solo-script/helloworld.py", line 7, in <module>
    vehicle = connect(target, wait_ready=False)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2775, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2048, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.
```
I hope this helps. Thanks!

hamishwillee, 2016-02-24:
Hi @reyes73a 

Your solo info output matches mine. When I install your tarball it installs, connects and runs fine. 

My "guess" is that some sort of local change has stuffed up your installation. About all I can suggest is trying to update to version 2.0.0. If that works then fine. If not, then consider doing a [factory reset](https://3dr.com/kb/factory-reset/) and then updating. Note that you'll need to use the "proper" version for updating on windows because of https://github.com/3drobotics/solo-cli/issues/41

reyes73a, 2016-02-24:
Let me try to update and see what happens. If it comes down to performing a
factory reset then my question would be what does a factory reset really
entail? Does it reset flight controller, onboard computer and remote
control?

Will have to start from step 1 in the dev guide and reinstall solo-cli and
all that correct?

Thank!

hamishwillee, 2016-02-24:
Yes, it resets everything on the solo - you will have to check that factory reset link to see what else it does. You won't need to reinstall solo-cli, but you will need to then update the software to the latest version, and you will *probably* need to log in at least once via SSH to your Solo before you will be able to connect using solo-cli

Another thing you might try *first( is doing ``solo resize``, which resizes your root partition. As part of this a lot of crud in your system may be removed (it was in my case) so that could help. Its worth trying before you go to the scary step of a factory reset.

alexblack, 2016-03-22:
I'm having this same problem, any solution?

```
ddc) Alexs-MacBook-Air:solo alex$ solo info
connecting to Solo and the Controller...
{
  "controller": {
    "ref": "3dr-controller-imx6solo-3dr-artoo-20160304215337",
    "version": "2.1.0"
  },
  "gimbal": {
    "connected": false
  },
  "pixhawk": {
    "apm_ref": "73e08529",
    "px4firmware_ref": "5e693274",
    "px4nuttx_ref": "d48fa307",
    "version": "1.2.23"
  },
  "solo": {
    "ref": "3dr-solo-imx6solo-3dr-1080p-20160304215558",
    "version": "2.1.0"
  }
}
(ddc) Alexs-MacBook-Air:solo alex$ solo script run hello_solo.py
connecting to solo...
checking pip...
script bundle already up to date.
running script...

Traceback (most recent call last):
  File "/log/solo-script/hello_solo.py", line 1, in <module>
    from dronekit import connect
ImportError: No module named dronekit
```

alexblack, 2016-03-22:
That was second try ever, first try looked different, but same error:

```
(ddc) Alexs-MacBook-Air:solo alex$ solo script run hello_solo.py
connecting to solo...
checking pip...
installing pip...  done.
installing wheel...  done.
installing virtualenv...  done.
uploading script bundle...
./
./hello_solo.py
./requirements.txt
./wheelhouse/
./wheelhouse/dronekit-2.4.0-cp27-none-any.whl
./wheelhouse/monotonic-0.6-cp27-none-any.whl
./wheelhouse/pymavlink-1.1.72-cp27-cp27m-macosx_10_11_x86_64.whl
Not deleting env/bin
New python executable in env/bin/python
Installing setuptools, pip, wheel...done.
cp: can't stat '/usr/lib/python2.7/site-packages/greenlet*': No such file or directory
Ignoring indexes: https://pypi.python.org/simple
pymavlink-1.1.72-cp27-cp27m-macosx_10_11_x86_64.whl is not a supported wheel on this platform.
running script...

Traceback (most recent call last):
  File "/log/solo-script/hello_solo.py", line 1, in <module>
    from dronekit import connect
ImportError: No module named dronekit
```