dronekit/dronekit-python #531: RC Override Question

dylan93, 2016-02-04:
I'm working on a project that wants to look into the affect of cyber attacks on UAV systems namely, their stability and mission performance. I want to simulate a cyber attack by setting a constant bias on achieved/actual rotor rate while the autopilot is commanding a different rotor rate. I came across the  [chanenel overrides](http://python.dronekit.io/examples/channel_overrides.html) which seems like it might be useful to me. My question is to ask if this is a viable approach for simulating a cyber attack on a UAV? Are there other options/avenues to pursue which would be easier/safer? I am not particularly familiar with the Dronekit API, but will be using it a lot this semester.

peterbarker, 2016-02-04:
On Wed, 3 Feb 2016, Dylan Thomas wrote:

> I'm working on a project that wants to look into the affect of cyber attacks
> on UAV systems namely, their stability and mission performance I want to
> simulate a cyber attack by setting a constant [...]

A "cyberattack" on a drone is unlikely to be this subtle.  What do you 
imagine the goal of an attacker here might be?  To steal the craft? 
Resetting the mission would probably be the easiest option if so.

You might consider an evaluation of the MAVLink 2.0 packet signing 
proposal as a means of mitigation, rather than looking at the current 
technology: 
http://lists.dronecode.org/pipermail/dronecode-tsc/2015-October/000171.html

Hopefully we will see packet signing within months.

dylan93, 2016-02-04:
So we want to purposefully alter the controllers output in order to simulate a "cyber attack", so I'm just trying to find the most sensible way to achieve this for now. Our research goal is to quantify the degradation of a "cyber attack" on a UAS and its hardware (e.g. how does it affect it's stability, controllability, mission performance, etc?). Also, we want to identify when cyber-physical degradations as they occur and then implement system ID to compensate, but that's another step in the future. I understand it's not a completely practical/feasible scenario, but there is vested interest in this project. Most research is being done in the identification and mitigation areas, so we are proposing to do physical testing and analysis on actual flight hardware.

That MAVLink 2.0 signing is definitely a helpful link for another side of the project (C2 link security).

peterbarker, 2016-02-04:
On Wed, 3 Feb 2016, Dylan Thomas wrote:

> So we want to purposefully alter the controllers output in order to simulate
> a "cyber attack", so I'm just trying to find the most sensible way to
> achieve this for now. Our research goal is to [...]

I guess my point is that its controllability and mission performance are 
not degraded at all - they are just under someone else's control...

stability will only be affected in certain modes and when the aircraft is 
pushed beyond its limits.

> identify when cyber-physical degradations as they occur and then implement
> system ID to compensate, but that's another step in the future. I understand
> it's not a completely practical/feasible scenario, but there is vested

It's not infeasible.  Using RC overrides is just not necessarily the way 
you would abuse any access you had to the aircraft.

> interest in this project. Most research is being done in the identification
> and mitigation areas, so we are proposing to do physical testing and
> analysis on actual flight hardware.

I assume, then, that you have an external system doing the monitoring, and 
that system knows roughly what the craft should be doing?

hamishwillee, 2016-02-04:
Or to put another way, the channel overrides are used to replace the values coming in from the RC signal, so any autopilot stabilisation etc that is being carried out (mode dependent) will still be carried out. 

So yes, this might be used to simulate "replacing the RC signal received by the vehicle" ... but you could argue just sending whatever signal you like from the RC transmitter would do exactly the same thing :-)

dylan93, 2016-02-04:
Ah, thank you both for the clarification on the channel overrides.

> Using RC overrides is just not necessarily the way
you would abuse any access you had to the aircraft.

What direction should I then take to achieve this?

> I assume, then, that you have an external system doing the monitoring, and
that system knows roughly what the craft should be doing?

Yes, so I was thinking you'd have a mission plan with an established flight path. At some point during the mission, the UAS experiences an "attack" causing a "cyber-physical degradation" e.g. change/control the bias on a single motor, so the actual rate and rate commanded by the Pixhawk are different (I do understand what you mean about actually just being controlled by someone else rather than being "degraded"). The external system maps/tracks/analyzes the effects of this "event" on the UAS's mission performance and controllability.

I hope that makes a bit more sense at least. Thanks for bearing with me on this, the project is still in it's infancy.

mikerob, 2016-02-04:
Aren't you really just doing external flight path analysis / validation? 

You don't need to alter anything on the pixhawk.  Just program your UAV to do something other than what your external system is programmed to expect, and see if it can tell that you're not following the "path".

hamishwillee, 2016-02-04:
I suggest perhaps you move this to a gitter channel for ardupilot or dronekit (https://gitter.im/dronekit/dronekit-python)

If you have a vehicle flying a mission and you send messages over its telemetry link you can do pretty much whatever you like since there is no way in MAVLink 1.0 to know that the messages are from a trusted source. So you could send messages to  change the RC values if you liked or you could delete the mission or change it. The autopilot is your slave - even if you have set it to obey a certain GCS by its system id, that ID can be set in your messages. 

-  I'm not sure how much power the transmitter has to lock you out by changing the mode to manual.
- If the original telemetry system is still in range it can send messages too - you can't stop it, though I guess you might be able to change the parameters so that it won't listen to the original GCS.
- You can't just modify one motor power because the autopilot doesn't expose single motors. You can change vehicle parameters so that the autopilot is misinterpreting what inputs are coming from where.

Anyway, again, given that there is no security for the messages to/from the vehicle, simulating an attack is just "sending commands". I guess you could also simulate the operator reacting, or as @mikerob suggests, see what happens if you send commands when the vehicle is under autonomous control.

hamishwillee, 2016-02-19:
I'm closing this question as "answered".

closed 2016-02-19 by hamishwillee