dronekit/dronekit-python #534: "Error running script unexpected token ' " when trying to using dronekit-python

makkaa94, 2016-02-06:
Hi there,

I am very new to this and am struggling! My current set up is too have an Ubuntu guest running the SITL and using the UDP port to communicate with mission planner on the windows host machine. I am writing code for a autonomous tracking mission for a university project and constructing the code in sublime text 2. I have installed dronekit-python every i could find in guides etc but when ever i try to 'import dronekit' or 'from dronekit import VehicleModes' and run the script in mission planner, an error of "Error running script unexpected token ' ". Am i using it in the correct way? are there any work arounds? Thank you very much for your time

Dan

mgrennan, 2016-02-07:
Im woking on the same thing.  Like you im having trouble learning all the connections between things.
Il post a video todat of what I have working and where im having troubles.my goal is to contribute to dronekit.

hamishwillee, 2016-02-08:
HI Dan

There is no reason why this should not work provided your Ubuntu guest is outputing UDP to a port that is visible to your host. Is there any particular reason why you're using SITL inside a Ubuntu guest - you can install it natively on Windows or even easier, use DroneKit-SITL.

Anyway, I have no reason why you're getting that warning unless you're actually including the `'` token in your text. 

What I would do is try to run the [vehicle state example](http://python.dronekit.io/examples/vehicle_state.html). That code is know to work - and the result you get from it will at least tell you if you have a good connection.

mgrennan, 2016-02-08:
Here is the Video I promised. - https://youtu.be/XRA3eqXzU98

This is demo of use the 3DR Dronekit-SITL.  In this video I connect MAVProxy, APMPlanner and then command it with simple-goto.py program in DroneKit-Python.

Dronekit-SITL is running on a Ubuntu remote laptop on IP 192.168.2.118.

MAVProxy and APMPlanner are running on a Windows laptop. This is where this video was made.

I then connect to a different window on the Ubuntu computer and connect the simple-goto.py program through the MAVProxy

DAN:
It took me some time to understand to understand how all the connections needed to be made.
Running Dronekit-SITL was simple enough and it was easy to understand it was opening up a TCP service on port 5760.  So connecting the simple-goto.py was just calling out that connection.  "--connect=tcp:192.168.2.118:5760"  
It was more difficult to figure out how to make MAVPoxy work.  The "--master=tcp:192.168.2.118:5760"  as straight forward. The trouble was calling out the UDP connections for other processes.  I had to learn "--out=udpout:192.168.2.118:1455" was a way to "OPEN" a connection to connection the simple-goto.py program to with the connection option "--connect=upd:192.168.2.118:14550" instead of the TCP connection. 

That lead me to understand how to connect APMPlanner at the same time with "--out=udpout:192.168.2.106:14550"

Part of the confusion is figuring out the --out=udpout command is "sending"  UPD packets at the receiving app when the app simple-goto.py is sending commands through MAVProxy to DroneKit-SITL.

mgrennan, 2016-02-08:
Also, I had trouble using Mission Planner 1.3.35.  Arming would cause a EFM error and just land again.

hamishwillee, 2016-02-08:
Thanks @mgrennan. Mavproxy is indeed the pain. There is some information on setup here: http://python.dronekit.io/develop/sitl_setup.html#connecting-to-dronekit-sitl
EFM or EKM error? Odd!

As an aside, I don't think this is @makkaa94 problem, but we won't know until we see if a known good example loads/runs without error (or with a different error).

makkaa94, 2016-02-08:
Thank you @mgrennan 

Hi Hamish, i was originally running it natively on windows, but after a months chirstmas break of not using it, it suddenly did not work anymore and there were discussions for potential fixes (due to a windows 10 update maybe) but i however could not fix it, so i decided to opt for a different method. 

I would just like to check my understanding of what i think should happen, here you'll see how little i know!

So after installing the dronekit-python packages, onto the host machine, and editing a code on the host machine, 'import dronekit' should give me access to a new set of commands to control the drone with?allowing me to construct a longer script, and being able to press play once and the drone will follow the script? 

I think now i will try the dronekit - SITL method.

again thank you for your time, much appreciated

hamishwillee, 2016-02-08:
@makkaa94 What you need to do is read the docs - that explains what dronekit-python will do for you and what dronekit-sitl will do for you: http://python.dronekit.io/ There are lots of examples. One to start with is in the [quickstart here](http://python.dronekit.io/guide/quick_start.html#basic-hello-drone) - this actually launches SITL for you from the script.

So Dronekit-Sitl is just an easy way of setting up SITL. It installs native binaries for your OS and runs them. This means that you don't need to build the source, or set up cygwin or anything like that. You have a version of SITL running on your machine, and you can connect to it on tcp:127.0.0.1:5760 from DroneKit. See http://python.dronekit.io/develop/sitl_setup.html

> So after installing the dronekit-python packages, onto the host machine, and editing a code on the host machine, 'import dronekit' should give me access to a new set of commands to control the drone [...]

After you install dronekit python you can call methods from the dronekit API. You can call `import dronekit` but it is more common to just import the bits you need (the minimal version is as below)

```
from dronekit import connect, 
```
You can construct long scripts or short scripts - check out the examples.
I don't know what you mean by "press play". You need to read the docs, because they really do give you step by step guidance. I am assuming you already have some idea of Python 2 programming.

Hope that helps!

makkaa94, 2016-02-08:
ah so i just set up the dronekit sitl, and had a simulation running from this. I then followed the quickstart, but an error popped up saying 'error running script no module named dronekit_sitl'. Is this an issue of not having the right folders in the right place? 

My project is for the drone to simply accept gps coordinates from an external source and the script to command the drone to follow these points iteratively, but must also include RC overrides and failsafes, and RTL when battery low etc. If i am unable to overcome these problems do you think this is possible without dronekit? 

You have been a huge help

hamishwillee, 2016-02-08:
Hi @makkaa94 

The error means that your python environment doesn't think dronekit_sitl is installed. I presume that you can get SITL running by entering the following on the command line?: 
```bash
dronekit-sitl copter
```
If that works, and if you can see dronekit-sitl in the list when you do `pip list` then probably you have multiple versions of Python installed.

Tell me if you get different behaviour if you start the **hello.py** using these two commands:
```
hello.py
```
or
```
python hello.py
```

makkaa94, 2016-02-09:
yes i do!

E:\Program Files\WinPython-64bit-2.7.10.3\python-2.7.10.amd64>hello.py
  File "E:\Program Files\WinPython-64bit-2.7.10.3\python-2.7.10.amd64\hello.py", line 1
    print 'hello world'
                      ^
SyntaxError: Missing parentheses in call to 'print'

E:\Program Files\WinPython-64bit-2.7.10.3\python-2.7.10.amd64>python hello.py
hello world

this is the different behavior.

peterbarker, 2016-02-09:
@makkaa94 You can add parentheses to the print() function to move forward while we continue investigating with you on this issue.  Here's some light reading: https://www.python.org/dev/peps/pep-3105/

hamishwillee, 2016-02-09:
@makkaa94 That is very indicative of multiple versions. You can get even better indication by running a script with this text in it using both ways (``script.py`` and ``python script.py``)
```
import sys
print sys.version_info
```
This will tell you the different versions being run. 

But I propose you ignore that, and instead run the original quickstart example using both approaches. ONE of them will be the environment that knows about your dronekit pip installation :-)

makkaa94, 2016-02-15:
Hi Hamish,

Still struggling with this! haha. So i have it all working through the WinPython command window, ie i can call on the example scripts and they work in the sitl. However my aim to to be able to select the scripts from within mission planner and run them from there on a real drone (i think this may have been poorly communicated by myself before). And this is where the problems start. When in Mission planner, i choose a script that works in the winpython command window, and click run script, and then am presented with the error 'Error running script No module named dronekit_sitl' or no module named dronekit for example. Is this a way dronekit can actaully work, by me calling a script to run and power a real drone directly through mission planner?

Thanks again

Dan

hamishwillee, 2016-02-16:
Hi Dan

I wish you'd made this clear before - that is a totally different problem. Essentially we have no idea what sort of environment mission planner uses for scripts - and the right way to add dronekit to it (it that is even supported).

There is information on Mission Planners (completely different) scripting approach here: http://planner.ardupilot.com/wiki/using-python-scripts-in-mission-planner/ . It looks like you could run dronekit in this environment too. I haven't tried it.

What I suggest you do is see if it works using those instructions above. If you run into any problems, then ask on the [Mission Planner Forum](http://ardupilot.com/forum/viewforum.php?f=12&sid=9105edd98e0caf58ee9dadc37b362ca6)

I am closing this, because I see integration with GCS as out of scope (certainly not a DK-Python "issue"). That said, I think it would make a very nice research project, so I've created a separate issue for that  here : #563

closed 2016-02-16 by hamishwillee