dronekit/dronekit-python #537: Dronekit delivery tracking needs to zoom and also ideally auto update

hamishwillee, 2016-02-08:
The track screen on the dronekit delivery example should ideally be updated to:

* Track the position of the vehicle without having to push the update button
* Support zoom
* Zoom initially to a reasonable level based on the distance between the initial position and the target position.

kosovit, 2016-07-25:
Hi @hamishwillee !
to make an autoupdate add change in track.html the following :
<head>
		{% include 'head.html' %}
	</head>
to 
<head>
		{% include 'head.html' %}
		 <meta http-equiv="refresh" content="1"> 
	</head>
This will make an auto refresh !

To make it support the zoom, I think there is a big work to do but you can modify the option zoom in the python file: change 'zoom': 13, to 'zoom': 19, . 19 is the maximun !

finally I have a question
since a week I did not see the map in the command and in the track . I think that there is a problem with 
access_token': 'pk.eyJ1Ijoia2V2aW4zZHIiLCJhIjoiY2lrOGoxN2s2MDJzYnR6a3drbTYwdGxmMiJ9.bv5u7QgmcJd6dZfLDGoykw',
            'mapid': 'kevin3dr.n56ffjoo',
I think I should change it , 
do you have the same issue ?
did you try to figure out ?

hamishwillee, 2016-07-25:
Hi @kosovit 

Those changes sound pretty reasonable to me, though we would also need to tidy up the documentation. 

How about you create a patch that we can review?

munkarkin96, 2017-01-28:
Hi @hamishwillee , 

If you don't mind me asking, how's the progress of this issue going? 

Thanks!

hamishwillee, 2017-01-30:
@munkarkin96 Not at all. This is more of an "enhancement", that we were hoping others would do.