dronekit/dronekit-python #556: Make all attributes read-only and have explicit setter methods

hamishwillee, 2016-02-11:
Most attributes are read-only but we have a number of writable attributes: ``Vehicle.mode``, ``Vehicle.armed ``, ``Vehicle.airspeed`` and ``Vehicle.groundspeed``, ``Vehicle.home_location``. 

There is nothing wrong with writable attributes "in general" but in this use case they aren't great because they imply attributes can be written synchronously and immediately ... and in our application they can't. This leads to "hey what" moments, means that we need to write somewhat complicated code to check the success of a write, and makes it easy to miss these checks.

The proposal is to make all the attributes read-only and move to explicit setters. The setters would be asynchronous, but provide wait_ready to allow synchronous use, timeout and callback for handling a fail case (with an exception being raised if there is no callback defined):
```
set_armed(wait_ready=False, timeout=10, callback=None)
```

So our current code:
```
    vehicle.armed = True    

    # Confirm vehicle armed before attempting to take off
    while not vehicle.armed:      
        print " Waiting for arming..."
        time.sleep(1)
```
would change to 
```
    vehicle.set_armed(wait_ready=True)  
```
or
```
vehicle.set_armed()  
    # Confirm vehicle armed before attempting to take off
    while not vehicle.armed:      
        print " Waiting for arming..."
        time.sleep(1)
```

Thoughts?

peterbarker, 2016-02-11:
If set_armed corresponds to is emitting a single "change mode" message, and that message gets lost, that last example is an infinite loop.  This is all through our example code.

It could be fixed by dropping vehicle.set_armed() into the loop - but its an easy pit to fall into.

peterbarker, 2016-02-11:
I would recommend having methods always named by what they do.  "set_armed" does not set the vehicle to armed.  If it sends a set_armed() message - call it send_message_set_armed().  "try_set_armed()" would be even better as it removes the details as to how the arming is achieved.

hamishwillee, 2016-02-11:
True. I've reworded this to capture your points.

hamishwillee, 2016-02-19:
This will be addressed by #566