dronekit/dronekit-python #563: Research Project: Integrate DroneKit-Python Scripts into Popular GSCs

hamishwillee, 2016-02-16:
Work out how to:

- run DK-Python scripts from popular ground stations
- deploy DK-Python scripts from popular ground stations to a companion computer and run them

Popular ground stations might include Mission Planner, APM Planner 2, Tower.

mikerob, 2016-02-16:
Running scripts would seem to require mods to the GCS itself, something like adding a console, or a "run script" function where you can set command line parameters. You'd need to pre-configure MAVproxy, or have a GCS that can output the MAVLINK stream. 

An interesting option would be enabling MAVLINK pass through in DK so you could use DK as your primary connection and connect the GCS to DK (you can do it now, but it requires MAVproxy to split / forward the connection)

Controlling scripts could be done using extra RC channels - set unused channels (12-18 for example) to high or low and have the DK script listen for those MAVLINK messages.

tcr3dr, 2016-02-17:
Ostensibly this should also be included: https://github.com/dronekit/tower-web

hamishwillee, 2016-02-18:
@tcr3dr Added tower web up the top. 

@mikerob I suspect you're right. Mission Planner has a scripting interface, and since that can import libraries I'm fairly sure we could do something with dronekit there. 

I don't think we'll enable MAVLink passthrough in DK. That would be reinventing the wheel of MAVProxy. We'd certainly consider it if the community put forward a strong use case that couldn't easily be met otherwise.

Research projects are by their nature, possibly impossible :-)