dronekit/dronekit-python #567: Channel Override needed? for forth and back Movement

Password2501, 2016-02-18:
Hi, 
Me and a Friend are working intensively on a Copter that is keeping it's distance from a wall,
As a distance measurement we do have a laser and a python-code that measures accurately. So our only option is to implement it with the channel override function.

Do you guys know a better way?
forth and back commands without using the GPS-coordinates

Thanks in Advance

hamishwillee, 2016-02-19:
I would at least try driving it using velocity commands as per http://python.dronekit.io/guide/copter/guided_mode.html#velocity-control

A better place to discuss this would be the ardupilot forums - because the best mechanism to use is defined in the autopilot and not dronekit.

Closing this here.

closed 2016-02-19 by hamishwillee

Password2501, 2016-02-22:
This wasn't very helpful...

hamishwillee, 2016-02-22:
I'm sorry it wasn't helpful. I don't know the answer; I've pointed you to the people who will be able to tell you whether velocity control is a reasonable option.