dronekit/dronekit-python #580: vehicle.commands.count reports 1 lower than it should

rmackay9, 2016-02-22:
when retrieving a list of mission commands, the count seems to always be one less than it should be.

So for example, if I load mission into Copter which is 2 waypoints, within ardupilot there will be 3 items in the command list: home, wp1, wp2.  It's arguably bad behaviour that command-0 is always home but this is besides the point.  From the dronekit user's point of view, when the command list is retrieved, there are actually 3 items that he/she can iterate over but the count is only 2.  this will lead people to not read the last command.

hamishwillee, 2016-02-22:
Hi @rmackay9 

That's because the mission commands on DKYP2.0 hides the existence of the home location in the 0th point. There are actually only however many points you get out of ``cmds.count``. The home location is separately set using an attribute. 

You can't actually access the cmds by index so cmds.count is pretty useless except for telling you when you have reached the last command via cmds.next.

I don't think there is any defect here, except that perhaps we should let people know that this differs from what they might be used to if using mavlink directly.

Thoughts?

rmackay9, 2016-02-22:
@hamishwillee,
I've reported to @peterbarker via Skype that the home_location always seems to be None so perhaps there's another issue as well.
It is possible to access the commands by index I think.  vehicle.commands[0] will return the first command.  I thought this was the home location.  In any case it seems to return a command so I think accessing commands by index is possible.

hamishwillee, 2016-02-22:
Hi @rmackay9 

home_location is None until the value is first set by the GPS - are you trying to read it before then? If you run https://github.com/dronekit/dronekit-python/blob/master/examples/mission_import_export/mission_import_export.py you can see that this does get the home location but it waits until the vehicle is armable before it does so to ensure there is a position.

YOu're right about the vehicle.commands[0] being accessible - indexed from 0 to commands.count-1. I think this is logical and consistent. Thoughts?

rmackay9, 2016-02-22:
I'm pretty sure the home position has been set but the vehicle.home-position is still none.  It seems strange to me that dronekit would wait for the vehicle to be is-armable.  Ardupilot sets the home position once the EKF is confident it has a good position but that doesn't mean the vehicle is armable.  It might not be armable for some other reason.  The two concepts should not be linked from within dronekit I think.
I'll check again if the 0th waypoint was home or not and reply again here.
Thanks for helping me with all this.  It's been a while since I've used dronekit.

rmackay9, 2016-02-22:
Ok, i'm really sorry.  It's me.  Dronekit is setting the vehicle.home_location once ardupilot has set it.  It seemed impossible to me that ardupilot hadn't set it's home location but in fact, what was happening is that the RPI2 was too close to the Pixhawk's GPS and it was absolutely killing the GPS signal.

So the point is that I should be using vehicle.home_location and not trying to use the mission list to get the home position.

Thanks and sorry!

closed 2016-02-22 by rmackay9

hamishwillee, 2016-02-22:
@rmackay9 Just FYI, and to be very precise, DK sets the home location as soon as it downloads a mission that contains a valid home location. 
1. We probably should now be calling [MAV_CMD_GET_HOME_POSITION](https://github.com/diydrones/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1445) - I assume this is in a released version (if so, do you know which it appeared in?)
2. The docs assume that a reader doesn't know that the mission could/should contain the home location. They are targeted at someone using DK but not knowing the underlying framework. If you think that we should clarify this for people like you who know how it really works, let me know :-)