dronekit/dronekit-python #602: Could not pip the simulator

anandbaskaran, 2016-03-04:
I am new to Drone kit and I am trying download the simulator by running `pip install dronekit-sitl`. But it says there exists some syntax error in the code. Please find the attached image for the detailed description.
[image]

peterbarker, 2016-03-04:
On Fri, 4 Mar 2016, Anand wrote:

> I am new to Drone kit and I am trying download the simulator by running pip
> install dronekit-sitl. But it says there exists some syntax error in the
> code. Please find the attached image for the detailed description.
> capture

That's a bug!

It has been fixed for several months in the master branch (see f9c84eff) - 
apparently it is past time for a release!

You can avoid this bug by using Python 2 rather than Python 3.

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

hamishwillee, 2016-03-06:
YOu can install from master using:

```
pip install -UI git+https://github.com/dronekit/dronekit-sitl
```

@peterbarker Are you making a new release?

anandbaskaran, 2016-03-15:
Even when I use Python2 I get a similar kind of error. I tried the command given bu hamishwillee too, but no progress :(

anandbaskaran, 2016-03-15:
Its working in master with Python 2 :) Thank you :+1:

closed 2016-03-15 by anandbaskaran