dronekit/dronekit-python #606: Using dronekit with native SITL on PX4

guachari, 2016-03-15:
I'm trying to use Dronekit and interact with native SITL by using the code given here : http://dev.px4.io/dronekit-example.html

However, the connection doesn't work. It keeps saying and then timeouts :

`>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition`

hamishwillee, 2016-03-15:
Thanks for the report. How are you setting up SITL - on what platform? Did you use the code exactly as in the example?

guachari, 2016-03-16:
Hi, 

I followed exactly the guide for Linux to install SITL as it is described here : http://dev.px4.io/starting-installing-linux.html
 I used this example : http://dev.px4.io/dronekit-example.html

I opened a conversation here, it seems that one person is having the same problem : https://groups.google.com/forum/#!topic/px4users/OY3Ly-tO69M

guachari, 2016-03-20:
Any ideas ?

mrpollo, 2016-04-12:
I'm sure this is an issue on our codebase, we are using some helper functions from pymavlink on mavlink/mavlink, I'm going to ask help upstream on PX4 Firmware

wookoouk, 2016-05-04:
I am having the same issue when connecting to a quad via mav

`vehicle = connect('/dev/tty.usbserial-DN009GXA', wait_ready=True)`

>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition

ndepal, 2016-06-05:
I am also trying to get DroneKit running with PX4 SITL. Installed the newest DroneKit version, have the newest version of the PX4 Firmware and am following the instructions given in http://dev.px4.io/dronekit-example.html. Like others, I am repeatedly getting the following error message:
```
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
```

Additionally, I occasionally see
```
>>> FREEFALL DETECTED
>>> LANDING DETECTED
>>> data link #1 lost
>>> data link #1 regained
```
All of these messages are also visible in the console where I am running the PX4 mainapp.
I'm not sure why I'm getting these freefall and landing messages, but that's a PX4 problem. But at least I see that DroneKit is getting messages from PX4.

What is the status on this issue? It doesn't look like a lot has happened since it was opened a couple of months ago. Is this being worked on?

DroneKit looks quite interesting, but with the current state it doesn't look like a usable tool for anybody developing with PX4.

samacumen, 2016-10-11:
Running the http://dev.px4.io/dronekit-example.html example..

$ sudo python px4_dronekit_eg.py -c "/dev/ttyACM0"

Connecting
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
^CTraceback (most recent call last):
  File "px4_dronekit_eg.py", line 40, in <module>

Seems like DroneKit python code is having a bug receiving the wrong ID for Mavlink command. Any solution/patch for this?

peterbarker, 2016-10-11:
On Tue, 11 Oct 2016, Shyam wrote:

> Running the http://dev.px4.io/dronekit-example.html example..
> 
> $ sudo python px4_dronekit_eg.py -c "/dev/ttyACM0"
> 
> Connecting
>
>                   Exception in message handler for
>                   HEARTBEAT
>                   mode 65536 not available on mavlink
>                   definition
>                   Exception in message handler for
>                   HEARTBEAT
>                   mode 65536 not available on mavlink
>                   definition
>                   Exception in message handler for
>                   HEARTBEAT
>                   mode 65536 not available on mavlink
>                   definition
>                   ^CTraceback (most recent call last):
>                   File "px4_dronekit_eg.py", line 40, in
> 
> Seems like DroneKit python code is having a bug receiving the wrong ID for
> Mavlink command. Any solution/patch for this?

This is typically because you're using an older pymavlink.

Repeatedly run "pip uninstall pymavlink" until it fails because it hasn't 
found one.  Then "MDEF=... python setup.py build install --user --force" 
in the pymavlink root.  Replace "..." with the path to your message 
definitions - typically a mavlink repository checkout.

samacumen, 2016-10-12:
I did the following on the PC I am running (Ubuntu) connected to Pixhawk:

$ sudo sudo pip uninstall pymavlink
## Successfully uninstalled pymavlink-2.0.6

(I only have dronekit on my PC, and by definitions I assume I must specify here, the dronekit path itself? Could you clarify?
$ sudo MDEF=dronekit python setup.py build install --user --force
## ...
## Finished processing dependencies for dronekit==2.9.0

$ python ../px4_dronekit_eg.py -c "/dev/ttyACM0"
## Connecting
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
...

Any inputs on the above? Possibly my path is incorrect? Thanks!

duboisj, 2016-11-19:
I am running the PX4 stack in a docker image as described in the PX4 devguide.  When I first tried dronecode I got the "mode 65536 not available on mavlink"  and HEARTBEAT errors described above.

I was able to fix this by installing a new pymavlink.

I did pip uninstall pymavlink until it complained that no pymavlink was found (this took 2 times for me).

Then I did:

git clone https://github.com/mavlink/mavlink
git submodule init
git submodule update
cd mavlink/pymavlink
python setup.py build
python setup.py install

After that, I re-ran my dronkit code and I was able to connect and retrieve basic vehicle information without error.

peterbarker, 2016-11-20:
@duboisj Many thanks for reporting that success!

@samacumen Please try @duboisj solution and reopen this issue if the same issue occurs.

closed 2016-11-20 by peterbarker

GregoireH, 2017-02-17:
Hello @peterbarker @duboisj 

I am also using PX4 Flight stack.
My goal is to connect from a computer to my drone indoor for running a script in order to have comparative results in logs.

I have been trying to clone mavlink and do the above command, to install the latest one.

My Pb is that, using dronekit 2.9.0, I then get this
```t
greg:vehicle_state$ python vehicle_state.py --connect '127.0.0.1:14550'

Connecting to vehicle on: 127.0.0.1:14550
>>> Link timeout, no heartbeat in last 5 seconds```

I also tried to connect to my usb directly with similar results.

If I downgrade dronekit down to 2.4.0 as at the top of this thread it doesn't work neither.

do you have any suggestions please ?
Thanks in advance for your help !

Best

IeiuniumLux, 2017-05-02:
I haven't used dronekit for python.  However, dronekit for android work well with SITL on PX4.  Here the result: https://youtu.be/TwOflR7KXuQ