dronekit/dronekit-python #607: Setting EXIF fields for lat/long/alt through the gimbal class

DarylWM, 2016-03-31:
For stitching images into maps, I need the current lat/long/alt injected into the EXIF header of each image. Is this likely to be a feature of the gimbal class?

franknitty69, 2016-03-31:
Daryl, exif data would have to be set by the GoPro. dronekit can change modes or take pictures, but it cannot alter the files on the gopro sd card. this is the reason that current method, even with other apm drones, is to write a log with this information and inject the information from the log into the images after downloading from the camera.

for dronekit to be able to set exif, dronekit would have to grab a frame from the current video stream, get the current lat/long/alt and write the image to solo with that exif info. this isn't the best way because the frame grab isn't going to best quality.

DarylWM, 2016-03-31:
Thanks Ahmed.

closed 2016-03-31 by DarylWM