dronekit/dronekit-python #608: Channels override

vinhk, 2016-04-01:
Hi my name is vinh k. I am working on an indoor vision based navigation in a GPS denied environment.  I want to be able to make quad go forward, backward, left,  and right.  Is channel override my only option? 

Thanks

manik9000, 2016-04-02:
I am also trying to do something like this.

mrpollo, 2016-04-12:
@rmackay9 can you help us clarify what's the recommended way here with APM?

alexblack, 2016-05-25:
Hi @vinhk any luck getting the drone to move with channel overrides?

vinhk, 2016-05-25:
I did not Alex. I have tried everything. :(

alexblack, 2016-05-25:
Frustrating! I haven't been able to find a single example of how to fly the
drone using channel overrides yet either. :(

vinhk, 2016-05-25:
Yes even the mavproxy rc 3 2000 does not do anything for me. I am expecting
this to be with firmware issues. However, with ar 3.2.1 you can at least
take off, hovering, and land using drone kit Python.

vinhk, 2016-05-25:
By the way for ar 3.2.1, use dronekit api Python on github v. 1.5 and not 2

vinhk, 2016-05-25:
https://github.com/dronekit/dronekit-python/tree/81bbd80076fb212c9305751333d9924e6b762434/examples

alexblack, 2016-05-25:
Have you managed to get the drone to takeoff with rc commands? Could you share some code?

vinhk, 2016-05-25:
I was not able to have it takeoff using rc override commands unfortunately.
I'm assuming this has to be with firmware issues where they expect you to
have gps navigation. Although optical flow would help but it's still
developmental I heard.

vinhk, 2016-05-25:
If you are able to use dronekit api Python with apm controller use this
code for takeoff. Be sure to fix the altitude. Right now it's at 20 meters

import time
from droneapi.lib import VehicleMode, Location
from pymavlink import mavutil

api = local_connect()
vehicle = api.get_vehicles()[0]

def arm_and_takeoff(aTargetAltitude):
    """
    Arms vehicle and fly to aTargetAltitude.
    """

    print "Basic pre-arm checks"
    # Don't let the user try to fly autopilot is booting
    if vehicle.mode.name == "INITIALISING":
        print "Waiting for vehicle to initialise"
        time.sleep(1)
    while vehicle.gps_0.fix_type < 2:
        print "Waiting for GPS...:", vehicle.gps_0.fix_type
        time.sleep(1)

    print "Arming motors"
    # Copter should arm in GUIDED mode
    vehicle.mode    = VehicleMode("GUIDED")
    vehicle.armed   = True
    vehicle.flush()

    while not vehicle.armed and not api.exit:
        print " Waiting for arming..."
        time.sleep(1)

    print "Taking off!"
    vehicle.commands.takeoff(aTargetAltitude) # Take off to target altitude
    vehicle.flush()

    # Wait until the vehicle reaches a safe height before processing
the goto (otherwise the command
    #  after Vehicle.commands.takeoff will execute immediately).
    while not api.exit:
        print " Altitude: ", vehicle.location.alt
        if vehicle.location.alt>=aTargetAltitude*0.95: #Just below
target, in case of undershoot.
            print "Reached target altitude"
            break;
        time.sleep(1)

arm_and_takeoff(20)

vinhk, 2016-05-25:
Before running this code, please download the entire folder from github
link I sent.

alexblack, 2016-05-25:
Thanks for the help, but, that code looks like it gets the drone to takeoff using normal dronekit methods.  I  was hoping to get the drone to takeoff with RC commands, like this: https://github.com/dronekit/dronekit-python/issues/629

vinhk, 2016-05-25:
That's my only option sadly. We have tried to imitate the rc commands by
using different frequencies for throttle, pitch etc but no prevail. I am
not sure why there is no documentation on rc overrides.

vinhk, 2016-05-25:
I'll also look into this issue as well so don't lose hope. You may find
solutions before me .cheers.

peterbarker, 2016-05-26:
On Wed, 25 May 2016, vinhk wrote:

> That's my only option sadly. We have tried to imitate the rc commands by
> using different frequencies for throttle, pitch etc but no prevail. I am
> not sure why there is no documentation on rc overrides.

There *is* the RC override example....

vinhk, 2016-05-26:
Yes, for v.2 of dronekit but since the takeoff commands did not work for
me. I just stuck with v.1.5 of dronekit. I was not able to have the
override work at all.

alexblack, 2016-05-26:
> There *is* the RC override example....

This one? http://python.dronekit.io/examples/channel_overrides.html

It doesn't seem to have any information about making the drone move.

alexblack, 2016-05-26:
hey @vinhk  I got the simulator drone (SITL) to takeoff with @peterbarker's help here: https://github.com/dronekit/dronekit-python/issues/629

vinhk, 2016-05-26:
I believed you needed to tried different pwm values for moving based on
channels.

vinhk, 2016-05-26:
Ok that's great .

rmackay9, 2016-05-26:
So really we should make a new mode called Guided-NoGPS in Copter that allows controlling the vehicle with attitude requests instead of using rc-overrides.  Allan Matthews implemented something like that but I haven't seen a PR pulling it in.
I'd really like to implement this, and I will eventually, but I've got a lot on my plate at the moment.  Basically the new flight mode should just be a very cut-down version of the existing Guided mode.. because Guided mode already responds to the set-attitude-target messages, it's just that Guided requires GPS when arming.

vinhk, 2016-05-26:
Yes randy can I help you with implementation. I would love to get involve
please let me know where can I begin. :)

alexblack, 2016-05-26:
Sounds like a great idea @rmackay9 , I don't quite follow though, wouldn't one also need to control throttle, not just attitude?

rmackay9, 2016-05-26:
@vinhk,
We have some instructions on the wiki here about adding new flight modes.  http://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html
They're a bit out of date but hopefully still useful.  The short form is that you'll need to create a new control_guidednogps.cpp file which should be a very cut-down version of control_guided.cpp.  It only needs to support one controller which is the guided_angle_control (i.e. "guided_angle_control_start", "guided_set_angle", "guided_angle_control_run").  So this new mode won't need most of the switch statements that control_guided.cpp has.  It would probably be good to do a search through the code to find other places where it checks "control_mode == GUIDED" to see if it also needs to check if "control_mode == GUIDED_NOGPS".
I'd recommend you join our jitter channel (if you're not already there) to get more real-time advice from me and the other devs.  guided_angle_control_run
Thanks!

vinhk, 2016-05-26:
Thanks Randy, I'll do as suggested.

alexblack, 2016-05-26:
hey @rmackay9 you mentioned being able to control attitude, how about throttle?

rmackay9, 2016-05-26:
@alexblack, so in regular Guided mode, when accepting this set-attitude-targets message and we interpret the thrust as meaning:
0.5 = stay at current altitude
1.0 = climb at WPNAV_SPEED_UP
0 = descend at WPNAV_SPEED_UP
So it's somewhat close to a percentage of the maximum possible climb rate.

alexblack, 2016-06-21:
I'm working with rc overrides, throttle control is working well with
channel 3.

What channels are roll and pitch? I've tried 1, 2 and 4, they don't seem to
have any effect.

(I'm putting the drone in STABILIZE mode)

alexblack, 2016-06-21:
Ok figured it out, pitch and roll are 1 and 2. Throttle seems to work well
with increments of 5, whereas pitch and roll need more like 100 to see any
difference

vinhk, 2016-06-21:
That's good news. What version of arducopter are you using? I gave up rc
override since it does not do anything to my motors.

alexblack, 2016-06-21:
I'm using the stock ardupilot that comes on the Solo.

vinhk, 2016-06-21:
I see. I'm wondering what version it is.

alexblack, 2016-06-21:
Here is a snippet of my code:

```
    """ Inspired by https://gist.github.com/vo/9362500#file-override_rc-py-L99 """

    self.vehicle.change_mode('STABILIZE')
    self.log(" Mode: {0}".format(self.vehicle.mode))

    self.log('RC mode started, waiting keypress...')
    self._log.flush()

    throttle = 1475
    roll = 1500

    while True:      
      k = read_single_keypress()
      if k == 'q' or ord(k) == 3:
        break 

      if ord(k) == 65: # up
        throttle += 5
      elif ord(k) == 66: #down
        throttle -= 5
      if ord(k) == 68: # left
        roll -= 100
      elif ord(k) == 67: #right
        roll += 100

      self.vehicle.channels.overrides[3] = throttle
      self.vehicle.channels.overrides[1] = roll
      
      self.log('Throttle: {}, roll: {}', self.vehicle.channels.overrides[3], self.vehicle.channels.overrides[1])
      self._log.flush()

      time.sleep(0.100)       

    # Reset 
    self.vehicle.channels.overrides = {}
    self.vehicle.change_mode('GUIDED')

# http://stackoverflow.com/questions/983354/how-do-i-make-python-to-wait-for-a-pressed-key
def read_single_keypress():
  """Waits for a single keypress on stdin.

  This is a silly function to call if you need to do it a lot because it has
  to store stdin's current setup, setup stdin for reading single keystrokes
  then read the single keystroke then revert stdin back after reading the
  keystroke.

  Returns the character of the key that was pressed (zero on
  KeyboardInterrupt which can happen when a signal gets handled)
  """
  fd = sys.stdin.fileno()
  # save old state
  flags_save = fcntl.fcntl(fd, fcntl.F_GETFL)
  attrs_save = termios.tcgetattr(fd)
  # make raw - the way to do this comes from the termios(3) man page.
  attrs = list(attrs_save) # copy the stored version to update
  # iflag
  attrs[0] &= ~(termios.IGNBRK | termios.BRKINT | termios.PARMRK 
                | termios.ISTRIP | termios.INLCR | termios. IGNCR 
                | termios.ICRNL | termios.IXON )
  # oflag
  attrs[1] &= ~termios.OPOST
  # cflag
  attrs[2] &= ~(termios.CSIZE | termios. PARENB)
  attrs[2] |= termios.CS8
  # lflag
  attrs[3] &= ~(termios.ECHONL | termios.ECHO | termios.ICANON
                | termios.ISIG | termios.IEXTEN)
  termios.tcsetattr(fd, termios.TCSANOW, attrs)
  # turn off non-blocking
  fcntl.fcntl(fd, fcntl.F_SETFL, flags_save & ~os.O_NONBLOCK)
  # read a single keystroke
  try:
    ret = sys.stdin.read(1) # returns a single character
  except KeyboardInterrupt: 
    ret = 0
  finally:
    # restore old state
    termios.tcsetattr(fd, termios.TCSAFLUSH, attrs_save)
    fcntl.fcntl(fd, fcntl.F_SETFL, flags_save)
  return ret
```

hamishwillee, 2016-06-21:
@alexblack FYI only, it is possible for the channels numbers to map to different things the default settings. I think RC_MAP is the parameter for that. I doubt it will matter to you, but figure you might want to know. The ardupilot.org wiki has information around channels.