dronekit/dronekit-python #609: any alternative for RC overrides?

manik9000, 2016-04-02:
this is what I am trying to do:
I am trying to write a python script which will control my drone (quadcopter + pixhawk).
but as a safety precaution I want to use joystick.
So, I will be setting a switch on joystick which should control/switch between joystick control and my script control.

can you suggest best method?
I am thinking to put my drone in LOITER mode as I just want to hover at height and detect an object.

mrpollo, 2016-04-12:
@rmackay9 do you have a preference here?

IMO if you are using joystick you RC overrides should be fine.

SpiderBall, 2016-06-14:
@mrpollo On the [Channel Overrides](http://python.dronekit.io/examples/channel_overrides.html) page it says that RC Overrides are not recommended unless they are being used for "certain types of joystick control." Could you please clarify this? I'm looking to implement a means of manually controlling my drone using an Xbox controller, and I'd like to verify if this is would be a proper use of RC overrides.

Thanks

mrpollo, 2016-06-14:
@SpiderBall that would most certainly fit the RC Overrides usage, this isn't really a restriction imposed by DroneKit you are free to use whatever method of controlling your drone you want, over at [APM Land](http://ardupilot.com/ardupilot/index.html) they have guidelines and recommendations on what to use to control your drone, they recommend RC Overrides for any manual input devices such as Joysticks or RC Controllers. Please add a new Issue if you are not satisfied by this response.

hamishwillee, 2016-06-14:
RC overrides are primarily intended for implementing software based joystick-like controllers (for example on-screen controllers on an app). It would certainly be OK to take the output from an xbox and then send it to the vehicle as an RC override but I have no idea how you'd do the "translation" from xbox to RC overrides in your dk-python app. 

A better approach might be to follow the [ArduPilot instructions on this topic](http://ardupilot.org/copter/docs/flying-with-a-joystickgamepad-instead-of-rc-controller.html#flying-with-a-joystickgamepad-instead-of-rc-controller) :-) 

I'm closing this issue as resolved.

closed 2016-06-14 by hamishwillee