dronekit/dronekit-python #610: Exception in message handler for heartbeat

dgorissen, 2016-04-03:
I found 3 threads about this but no authorative answer/solution:

https://groups.google.com/forum/#!topic/drones-discuss/tQQjc0_5Eng
https://discuss.dronekit.io/t/issue-with-vehicle-state-py-and-rover/307
https://discuss.dronekit.io/t/missionplanner-and-dronekit-floods-console-with-errors/287

Im running dronekit 2.4.0 on an odroid UX4, connected with an FTDI cable to the Pixhawk Telem 2 port (dronekit is reading straight from /dev/ttyUSB0). Pixhawk is running APM copter 3.3.3 and the 3DR Radio is on Telem 1.  Serial 4/5 is also in use (Taranis telemetry).

I notice the following messages being spammed to std err/out on the odroid. Otherwise it does seem to be working properly (both reading the data via dronekit and apm planner 2.0).

```
...
argument of type 'NoneType' is not iterable
Exception in message handler for HEARTBEAT
argument of type 'NoneType' is not iterable
Exception in message handler for HEARTBEAT
argument of type 'NoneType' is not iterable
Exception in message handler for HEARTBEAT
...
```

It seems the issue is something to do with the telem1 & telem2 streams interfering with each other somehow? Some guidance as to what the correct usage pattern is would be appreciated. Or confirm if this is a bug.

PureRockets, 2016-04-03:
I am having the same error, however it is critical for me.  I'm using an ErleBrain2 running APM. It always returns Not Armable even when it's armed. When I try to run vehicle_state, I get several of these heartbeat exceptions, then it crashes (same as reported in 307)

peterbarker, 2016-04-04:
Attempting to iterate over NoneType is clearly a bug.  We'll need more debug to try to track this.

Could you add the following lines:

                import traceback
                traceback.print_exc()

into mavlink.py, just under this line:
                                errprinter('>>> ' + str(e))

please?

dgorissen, 2016-04-04:
It was actually in a slightly different place that the edit was needed but none the less:

```
>>> Exception in message handler for HEARTBEAT
>>> argument of type 'NoneType' is not iterable
Traceback (most recent call last):
  File "/home/dgorissen/.virtualenvs/orangspy/local/lib/python2.7/site-packages/dronekit/mavlink.py", line 150, in mavlink_thread_in
    fn(self, msg)
  File "/home/dgorissen/.virtualenvs/orangspy/local/lib/python2.7/site-packages/dronekit/__init__.py", line 1011, in listener
    self.notify_message_listeners(msg.get_type(), msg)
  File "/home/dgorissen/.virtualenvs/orangspy/local/lib/python2.7/site-packages/dronekit/__init__.py", line 1467, in notify_message_listeners
    fn(self, name, msg)
  File "/home/dgorissen/.virtualenvs/orangspy/local/lib/python2.7/site-packages/dronekit/__init__.py", line 1169, in listener
    if self._is_mode_available(m.custom_mode) == False:
  File "/home/dgorissen/.virtualenvs/orangspy/local/lib/python2.7/site-packages/dronekit/__init__.py", line 1501, in _is_mode_available
    return mode_code in self._mode_mapping_bynumber
TypeError: argument of type 'NoneType' is not iterable
```

fnoop, 2016-04-04:
I've added a small PR to deal with the specific iterable problem, but it uncovers the underlying problem which I think is the GCS (or whatever) is using vehicle mode 0 which isn't in the mode mappings.   This can be dealt with in several places:
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition

fnoop, 2016-04-04:
So the exception in the message handler should be dealt with better, but the underlying issue is almost certainly related to this:
https://github.com/dronekit/dronekit-python/issues/516
The connection the dronekit script makes should be source filtered so it doesn't receive messages from the entire mavlink network.

PureRockets, 2016-04-04:
I get the exception with a single vehicle, and just dronekit access (no Mission Planner). I'm dead in the water with dronekit, I can't arm my vehicle.

peterbarker, 2016-04-04:
@fnoop https://github.com/dronekit/dronekit-python/pull/585

dgorissen, 2016-04-05:
@fnoop, @peterbarker thanks. Looking at #585 it hasnt seen updates for a couple months (other than the CI failing). Im currently staying on the safe side and plugging in only a single telemetry consumer (my companion computer). Anything else and I get crosstalk. This means I cannot use my radios, taranis telemetry, etc. which is not great as Im flying blind.

From the issue it seems like a hairy one and will take some time to resolve (?). Any suggested workarounds or fixes?

mrpollo, 2016-04-12:
@dgorissen are you running APM or PX4?

dgorissen, 2016-04-12:
APM

mrpollo, 2016-04-25:
@dgorissen this was fixed in #611 release to follow shortly

dgorissen, 2016-04-25:
Awesome. Thanks @mrpollo

hannahdn, 2016-06-22:
I am still getting this error (after #611 update):
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition

@fnoop You mentioned a fix for that, what do you use?

wookoouk, 2016-08-02:
with everything up-to-date i still get:

>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
c>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition

squilter, 2016-08-02:
@ wookoouk APM or px4?

fnoop, 2016-08-03:
@hannahdn @wookoouk - the underlying issue is still there, dronekit doesn't handle multiple sources on a link properly.
I'm wondering if dronekit has been somewhat given up on for things like this - I notice that recent 3dr swarm work has been done with ROS.

peterbarker, 2016-08-04:
@fnoop @hannahdn @wookoouk fnoop is correct that dronekit-python doesn't handle multiple vehicles on one link properly at the moment.  The problem is primarily in pymavlink, 'though, not dronekit-python.

I have had a great deal of success with my avoidance work and dronekit-python, however.  Very reliable, multi-vehicle testing to the point I was happy to call the suite a regression suite.

https://github.com/peterbarker/dronekit-python/tree/source-system-filtering/examples/avoidance

(those instructions are a little out-of-date; the avoidance work in ardupilot is upstream now)

expntly, 2016-08-29:
@peterbarker 
@LorenzMeier 

Please kindly update on this issue.  Here's all the info I have collected after a few hours of debugging, hope this helps and let me know if/how I can help.

I'm running a Pixhawk (HKPilot32 really) with the PX4 flight stack.  It doesn't look like it's playing well with mavlink proto definitions.  This is preventing me from using Dronekit at all.  While debugging I should mention I tried ArduPlane 3.6.0 which works just fine.  Using QGroundControl as a GCS.

```
$ mavproxy.py --master=/dev/serial0 --baudrate 57600 --out udp:127.0.0.1:14550
Connect /dev/serial0 source_system=255
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from /dev/serial0
online system 1
ARMED
Mode(100925440)> Mode Mode(100925440)
fence breach
APM: data link #1 regained

Mode(100925440)> 
Mode(100925440)> mode stabilize
Mode(100925440)> Got MAVLink msg: COMMAND_ACK {command : 176, result : 0}
APM: [cmd] DISARMED by set mode command
APM: failsafe mode off
DISARMED
Mode(65536)> Mode Mode(65536)

Mode(65536)> 
Mode(65536)> APM: [blackbox] stopped (50146 drops)

```

Trying to connect from another console and run vehicle_state.py pointing at mavproxy:

```
$ python vehicle_state.py --connect "udp:127.0.0.1:14550"

Connecting to vehicle on: udp:127.0.0.1:14550
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
```

Killing mavproxy (and QGroundControl) and running vehicle_state.py directly pointed at the serial port (this is a companion computer setup):

```
$ python vehicle_state.py --connect "/dev/serial0"

Connecting to vehicle on: /dev/serial0
>>> Link timeout, no heartbeat in last 5 seconds
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "vehicle_state.py", line 37, in <module>
    vehicle = connect(connection_string, wait_ready=True, rate=57600)
  File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 2827, in connect
  File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 2100, in initialize
dronekit.APIException: Timeout in initializing connection.
```

Version info here:

```
$ pip list | grep mav
pymavlink (2.0.6)
$ pip list | grep drone
dronekit (2.8.0)
dronekit-sitl (3.1.0)
```

I tried multiple combinations of pymavlink and dronekit versions, without success.

Also tried all PX4 firmware versions (stable, beta, master as of today Aug 28, 2016), which doesn't help.

peterbarker, 2016-08-29:
@expntly You will need to use the version of pymavlink found in the "mavlink" repo on github (https://github.com/mavlink/mavlink).  This is typically done with "python setup.py build install --force --user" in the pymavlink directory.

expntly, 2016-08-29:
@peterbarker Indeed that did the trick thanks!  Any chance a mavlink release can be pushed?  Or at least a warning message somewhere.  That would have saved hours of frustration :-)

LorenzMeier, 2016-08-29:
@mrpollo @peterbarker This has been fixed for PX4 a very long time ago and it would be nice to finally have a release addressing it.

mrpollo, 2016-08-29:
Thanks @LorenzMeier I'll cut a new release asap.

mrpollo, 2016-08-30:
I released 2.9.0 yesterday, can someone please verify this is fixed on that release?

kostkar, 2016-09-03:
Hello all, I tried it with dronekit 2.9.0 and both pymavlink 2.0.3 and the mavlink/pymavlink(https://github.com/mavlink/mavlink) in both cases I see the >>> mode 0 not available on mavlink definition, as well as intermittent connection. Is there any plans for fixing this, I am available to test it, and flight test it as soon as it seems safe :).

majorpr13, 2016-09-18:
Has there been any luck on this...I am having the same issue.

paan2097, 2016-09-20:
I'm also getting the mode 0 heartbeat error that prints out in my dronekit terminal whenever I connect with Mission Planner.

expntly, 2016-09-26:
Last time I had it a few days ago, it was with the following setup:
 - RPi connected to Pixhawk via Telem 2
 - Mission Planner connected to Pixhawk via 915Mhz telemetry Telem 1

I tried running dronekit from the Pi and got the "mode 0 not available on
mavlink definition" exception message.  The moment I disconnected MP, the
issue went away.

So the bottom line is that there seems to be some weird interaction when
multiple mavlink connections are made to the Pixhawk.

Corusco, 2016-09-29:
I'm seeing the "mode 0 not available" error as well. I'm on PX4 flight stack, connecting my mac over FTDI cable to pixhawk Telem2.

I was getting a mixture of mode 0 and mode 65536 until I updated my pymavlink. I was able to get data down to populate some properties, but heartbeat was never consistent. Now it's the heartbeat error all the time.

Even if this isn't completely fixed in dronekit, I'm all about workarounds right now. I have some code I desperately need to ship.

majorpr13, 2016-09-29:
@Corusco This was actually fixed in another github issue. First...run this command repeatedly until pip reports pymavlink is not found or no longer installed. pip uninstall pymavlink

Then follow these instructions 
1) git clone https://github.com/Ardupilot/pymavlink
2) cd into the pymavlink directory
3) python setup.py build install --force --user

this should install the latest. There has been an issue when following multiple sets of instructions for the installs and how pymavlink installed from certain sources is updated with the latest.

majorpr13, 2016-09-29:
@Corusco You can read more about it over here https://github.com/dronekit/dronekit-python/issues/632#event-797047749

expntly, 2016-09-29:
Just curious, what's the difference between Ardupilot/pymavlink and mavlink/mavlink?  Which one is preferred?

paan2097, 2016-09-30:
@majorpr13 When I run those commands, after the third command it returns No XML message definitions found. This is suppose to fix the error when having a dronekit and mission planner both connected to a Pixhawk/SITL right??

majorpr13, 2016-09-30:
@paan2097 Did you continue to execute pip uninstall pymavlink until it reported there were none left? Did you cd into the correct directory that was cloned from git? Which version of python are you running? It may be easiest if you direct message me a screenshot of your console on all of the steps.

salonijain12, 2016-10-03:
We are trying to connect rpi with pixhawk serially on Telem2 using the link: http://plane.ardupilot.com/dev/docs/raspberry-pi-via-mavlink.html
Able to connect MAVProxy on Pixhawk from Rpi but when using it with dronekit-python (running the python script), getting the following error :
-Exception in message handler for HEARTBEAT
-mode 65536 not available on mavlink definition
 
Following are the versions used:
**PX4 stack firmware 1.4.4** installed on pixhawk.
**Dronekit-python==2.9.0**, installed from github source : https://github.com/dronekit/dronekit-python.git
**MAVProxy==1.5.4** (also tried with **MAVProxy==1.5.1**)
**pymavlink==2.0.6**

Can anyone please suggest if something is being missed?
What is the correct combination of dronekit-python, MAVProxy and pymavlink versions for rpi and pixhawk connection.

lroberts99, 2016-10-04:
@majorpr13 I am also facing the same issue of 
` Exception in message handler for HEARTBEAT 
 mode 0 not available on mavlink definition`

I followed all of your steps but I am facing the same No XML definitions found message as @paan2097 . Any ideas?

majorpr13, 2016-10-04:
@paan2097 @salonijain12 @lroberts99 
You need to be more clear when asking questions about what you are actually doing...settings firmware etc. Everybodys scenario may be entirely different and without a good background its hard to debug online. That being said, this thread was started via the PX4 running the ardupilot based firmware not the PX4 based firmware. I have not tested anything using the PX4 variant SW base. 
If I had to guess it cant find the directory containing the XML definitions because you didnt have permission for the install. Try the uninstall process again and then you could take a shortcut and try sudo pip install pymavlink.

lroberts99, 2016-10-04:
Sure, sorry about that. I am using running the dronekit commands from an Edison board on a Solo drone. I have these connected via USB. I ended up installing dronekit this way http://stackoverflow.com/questions/35964326/raspberry-pi-proper-connection-to-pixhawk-with-dronekit. I still get the exception, but it only prints out twice and then moves on to the rest of my code. Thanks for your reply.

Corusco, 2016-10-04:
@majorpr13 Apologies for not thanking you last week for your help. I no longer get the Mode 65536 error (though Mode 0 still pops up), so I've seen improvement there. 

@lroberts99 @paan2097 I experienced the missing XML definitions problem when pymavlink was installed in a directory outside of the Mavlink/pymavlink directory, where it should sit. I would suggest reinstalling mavlink by cloning from https://github.com/mavlink/mavlink, and updating the submodules or navigating into the pymavlink directory (inside the mavlink directory) and cloning from https://github.com/ArduPilot/pymavlink. Running the setup should complete successfully here.

majorpr13, 2016-10-04:
@Corusco when you connect are you connecting with mavproxy or dronekit?

Corusco, 2016-10-04:
@majorpr13 That was with the dronekit vehicle_status example. Mavproxy wasn't returning anything at that point. I've since moved to using a Jetson TK1 as my companion computer. Now, the example doesn't return anything, but MavProxy is returning a lot of garbled data along with some of the message headers interspersed. I'm wondering if my problem there was a configuration problem or a Mac problem. I did fully reinstall mavlink, dronekit, and pymavlink on that setup. Working on debugging the TK1 now. So, my current status is not entirely related to this issue, but I could step back on to my mac to push through if we want to address a bug.

majorpr13, 2016-10-04:
@Corusco what port are you connected on the TK1....are you using the correct voltage logic levels for comms between the devices?

Corusco, 2016-10-04:
@majorpr13 I did start with the UART2 on the GPIO pins over a line level shifter. I had a lot of noise in that configuration, so I'm currently using USB-TTL FTDI cable.

majorpr13, 2016-10-04:
@Corusco thats strange. I have my pixhawk running through the GPIO of my odroid and dont have a problem and thats a level shift from 1.8 to 3.3. Do you have the correct TTL level...of course if its the same used with your mac you should get the same results. i think about it let me know if you get it...curious problem

Corusco, 2016-10-04:
@majorpr13 Yeah, I also had 1.8 to 3.3 with the TK1, using the TI TXB series shifters. I tried three chips to be sure it wasn't a chip problem, but all three exhibited the same noise. 

A couple other local guys are also using odroid problem-free, but we're locked in to the tegra chip, so a platform switch isn't an option. But I know several others have also used the TK1 successfully.

In terms of the USB-TTL cable right now, it's definitely a 3.3v.

Thanks again for being willing to help me out with this. I'll let you know if I find anything.

lidia-toscano, 2016-10-06:
Hi, I am having the same issue and I've tried following the instructions:
Then follow these instructions 
1) git clone https://github.com/Ardupilot/pymavlink
2) cd into the pymavlink directory
3) python setup.py build install --force --user

However when I try to do step #3 I get the following error:
root@beaglebone:~/pymavlink# python setup.py build install --force --user
No XML message definitions found

Any ideas what I need to do to get the last step implemented?

Thanks in advance.

lidia-toscano, 2016-10-06:
Now the problem is even worse because when I try to execute my program I get this:

root@beaglebone:~# python basic.py
WARNING: Unable to load XML validator libraries. XML validation will not be performed
Traceback (most recent call last):
  File "basic.py", line 6, in <module>
    from dronekit import connect, VehicleMode, LocationGlobalRelative, LocationGlobal
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 42, in <module>
    from pymavlink import mavutil, mavwp
  File "/root/pymavlink/mavutil.py", line 105, in <module>
    set_dialect(os.environ['MAVLINK_DIALECT'])
  File "/root/pymavlink/mavutil.py", line 96, in set_dialect
    mavgen_python_dialect(dialect, wire_protocol)
  File "/root/pymavlink/generator/mavgen.py", line 197, in mavgen_python_dialect
    if not mavgen(opts, [xml]):
  File "/root/pymavlink/generator/mavgen.py", line 82, in mavgen
    xml.append(mavparse.MAVXML(fname, opts.wire_protocol))
  File "/root/pymavlink/generator/mavparse.py", line 278, in __init__
    f = open(filename, mode='rb')
IOError: [Errno 2] No such file or directory: 'message_definitions/v1.0/ardupilotmega.xml'
root@beaglebone:~# ^C

Any ideas?  

Thanks

Corusco, 2016-10-06:
@lidia-toscano In my experience, the XML definitions pymavlink is looking for are located in the Mavlink library, unless you generate your own and give pymavlink a reference to their location. The easiest and most thorough solution that I have found is to install mavlink by cloning from https://github.com/mavlink/mavlink, and updating the submodules or navigating into the pymavlink directory (inside the mavlink directory) and cloning from https://github.com/ArduPilot/pymavlink. Running the setup should complete successfully here.

peterbarker, 2016-10-06:
On Thu, 6 Oct 2016, lidia-toscano wrote:

> Hi, I am having the same issue and I've tried following the instructions:
> Then follow these instructions
> 1) git clone https://github.com/Ardupilot/pymavlink
> 2) cd into the pymavlink directory
> 3) python setup.py build install --force --user

The pymavlink repo doesn't contain the message definitions, but does 
include the code to generate language binding from the message 
definitions.  You may specify the location of the message definitions 
using the "MDEF" environment variable.

e.g.
pbarker@bluebottle:~/rc/pymavlink(master)$ 
MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions python 
setup.py  build install --user --force

lidia-toscano, 2016-10-07:
Thanks for your responses.  Corrusco, I followed your directions and I think the clone was installed correctly.  At any rate, now I have a directory for mavlink and a subdirectory within mavlink with a pymavlink clone full of files.  I do see the setup.py file in there.  Peter is right in that the XML files are not where they should be, during the cloning there were notices that they would not work.  I then tried to follow the last instruction provided by Peter but it is giving me the following error:

root@beaglebone:~# /rc/pymavlink(master)$MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions python setup.py  build install --user --force
-bash: syntax error near unexpected token `master'
root@beaglebone:~#

Any other ideas on how to correct this error?

Thanks again,

Lidia

majorpr13, 2016-10-07:
What is this command doing? /rc/pymavlink(master)$MDEF=

majorpr13, 2016-10-07:
That is what its complaining about

lidia-toscano, 2016-10-07:
The pymavlink repo doesn't contain the message definitions, but does
include the code to generate language binding from the message
definitions.  You may specify the location of the message definitions
using the "MDEF" environment variable.

e.g.
pbarker@bluebottle:~/rc/pymavlink(master)$
MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions python
setup.py  build install --user --force

lidia-toscano, 2016-10-07:
I found where the definitions are located.  What is the procedure to
include them in my directory?  Can I clone them in there?  Thanks, I am a
real newbie.

https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml

Thanks a bunch!

peterbarker, 2016-10-09:
On Fri, 7 Oct 2016, Lidia wrote:

> files. I do see the setup.py file in there. Peter is right in that the XML
> files are not where they should be, during the cloning there were notices

> root@beaglebone:~#/rc/pymavlink(master)$MDEF=$HOME/rc/ardupilot/modules/mavlink/message_defin
> itions python setup.py build install --user --force
> -bash: syntax error near unexpected token `master'
> root@beaglebone:~#
> 
> Any other ideas on how to correct this error?

Sorry, I was unclear in my instructions.  What I pased in was an excerpt 
from my shell, supposedly giving context to the command line I was 
suggesting you run, but apparently confusing you!

The command you should run is:

MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions python setup.py build install --user --force

And you should run that in the "pymavlink" github clone's directory.  You 
may need to adjust the location of the directory - in this case I'm using 
the definitions out of the mavlink submodule of an ardupilot repository.

This command line is setting an environment variable before running 
python.

> Lidia

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

paan2097, 2016-10-10:
I successfully ran this command inside the pymavlink directory: MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions python setup.py build install --user --force

Unfortunately, when I run a SITL or try to connect both a dronekit script along with mission planner I am still receiving this message:
Exception in message handler for HEARTBEAT mode 0 not available on mavlink definition

lidia-toscano, 2016-10-10:
Peter,

Thank you so much for your suggestions.  I did NOT successfully run the
code indicated:  MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions
python setup.py build install --user --force.  I adjusted the directory to
fit my setup thus:

 MDEF=$HOME/rc/mavlink/pymavlink/message_definitions python setup.py build
install --user --force   (because I do not have an ardupilot and modules
folders).  Here is my dilemma, I don'r know enough to move folders around.
I might cause more damage than I intend to so far.

My mavlink folder HAS a "message-definitions" folder which for some reason
Dronekit does not see and therefore gives me the mode 0 errors.  You
indicated to run the
 MDEF=$HOME/rc/mavlink/pymavlink/message_definitions python setup.py build
install --user --force command in the pymavlink folder which is inside the
mavlink folder.  There is NO setup.py file in the mavlink folder.  It is
buried in the pymavlink so when I run the

MDEF=$HOME/rc/mavlink/pymavlink/message_definitions python setup.py build
install --user --force  so it cannot see the "message_definitions folder.

root@beaglebone:~# cd mavlink
root@beaglebone:~/mavlink#
MDEF=$HOME/rc/mavlink/pymavlink/message_definitions python setup.py build
install --user --force
python: can't open file 'setup.py': [Errno 2] No such file or directory
root@beaglebone:~/mavlink# cd pymavlink
root@beaglebone:~/mavlink/pymavlink#
MDEF=$HOME/rc/mavlink/pymavlink/message_definitions python setup.py build
install --user --force
No XML message definitions found
root@beaglebone:~/mavlink/pymavlink#

Can I just move the message_definitions folder down to pymavlink without
breaking anything?

Thanks!

lidia-toscano, 2016-10-10:
I got brave and moved it.  Still can't get it to work!

paan2097, 2016-10-10:
@lidia-toscano Are you sure you have your directory path set up right? My ardupilot directory is located directly in my home folder, so I had MDEF=$HOME/ardupilot....

lidia-toscano, 2016-10-10:
Paan2097,

Not really sure - I am logged in as root@beaglebone

Is that the "HOME" or is "HOME" another command like MDEF?  I am not sure
what those commands are saying, like MDEF, HOME, rc etcetera.  This is all
new to me and I am learning but it seems I have to google just about
everything.  I'm having a bit of a learning curve.  lol

paan2097, 2016-10-10:
@lidia-toscano Welcome to the world of programming. HOME is the base directory. When you are log in as root@beaglebone, you start at the root directory. A few directories are built in, like HOME/Desktop, /Documents, and /Downloads. So it depends on where you cloned the ardupilot directory. Where in your folder system in the ardupilot folder?

[16 more replies]