dronekit/dronekit-python #613: Chinese team will start to translate this docs very soon

bys1123, 2016-04-07:
We got two members good at Dronekit.

zlite, 2016-04-07:
Great news. I'm really pleased to see this effort in China.

Chris

hamishwillee, 2016-04-07:
Fantastic! This is a very similar infrastructure to the ardupilot wiki (only the theme is different) so we can use the same approach we're working on for that.

Thanks Stone!

mrpollo, 2016-04-12:
@bys1123 is there anything we can help you guys to get this moving?, also really interested to know about your team, please send a pull request as a WIP with your branch changes, it's good to have early feedback to avoid big reviews

bys1123, 2016-04-13:
@mrpollo 
We translate Sphinx wiki like this way: https://github.com/ArduPilot/ardupilot_wiki/issues/355
And we will creat a new github project to manage chinese translation files.
Now I have a problem, I don't know why a sentence will break into different line, when I update the files to Transifex, it will like this: https://github.com/ArduPilot/ardupilot_wiki/issues/363
it's hard to translate because of sentence are all aparted.
Are you using Emacs or something? Can i fix that with use some software?
@hamishwillee

bys1123, 2016-04-13:
@mrpollo I'm not good to speak english, so my words seems rude sometime. but I'm really thanks you guys for helping me. I'll keep learning.

bys1123, 2016-04-13:
@mrpollo you could know something me from this article: https://groups.google.com/forum/#!topic/drones-discuss/tAMH5o_dbGc

mrpollo, 2016-04-13:
@bys1123 you are free to modify the CR-LF line breaks if you need to but I would prefer @hamishwillee guidance here since you already faced this issues on ardupilot

@bys1123 I think your messages get across perfectly fine, you don't need to worry about the tone of the conversation.

hamishwillee, 2016-04-14:
Howdy both. 

As per https://github.com/ArduPilot/ardupilot_wiki/issues/363 (I just responded) Sphinx doesn't care about having broken up lines, and having them makes it (very slightly) easier to see changes on github when pages are updated.

I think we should remove the lines to ease translation, remembering of course that someone else may add them in at some time ... lots of people working on content hopefully.

I do not believe there is any automated way to do this safely - in particular you'll have problems with tables etc. If you want to do this, it will have to be by manually going through docs and submitting patches. I'm happy to do documents as I work on them, but I don't currently have time to do a full sweep of all our docs.

@mrpollo Just FYI, sphinx has its own translation mechanism where you can dump out "PO" files, translate them, and then build a specific version using these translations + the theme translation strings. These translation strings can live in the git tree and be updated as per any other file (a slightly complicated process as strings change and you need to keep everything updated). Stone's team are using transifex as an intermediate tool to mediate the translation. This appears to work well, and I am hoping to try it out sometime. They will share their translations with us after they've got a bit further along.

Also FYI Transifex supports another model for translation "transifex live" where Transifex detects strings/changes and you can edit translation in place on your website - javascript is used to select the right translation to display. I used this on the old wordpress ardupilot site - awesome but a bit of a maintenance pain for me and also you lose some features like ability to search in the language. Unfortunately this service is slow to edit and use on the other side of the "great wall". I'm in two minds about what to do on this side of the great wall for other languages - and I won't know until I've done some more work on the solution Stone is using.

Anyway, Stone, does it make sense for you to manually remove CR+LF in places it is breaking translation?

As to how