dronekit/dronekit-python #617: Cannot arm with Channel Overrides- stuck at Initialising APM

jainakshay91, 2016-04-19:
I have been trying to arm my quad with RC Overrides, but I cannot do so as it gets stuck at Initialising APM and then jumps to the next statement in my code. Any ideas regarding this issue?

Thanks in Advance, 
Best 
Akshay

hamishwillee, 2016-04-19:
Please create a gist with your code and link to it from here.

jainakshay91, 2016-04-19:
[Link to my code](https://gist.github.com/jainakshay91/1b44a41cd63728ca65f24c5e452c5511)

hamishwillee, 2016-04-19:
I'd re-write this to run on SITL - makes testing cycle a lot easier. Is there any reason you can't use conventional arming code for the takeoff?:

```
    print "Basic pre-arm checks"
    # Don't try to arm until autopilot is ready
    while not vehicle.is_armable:
        print " Waiting for vehicle to initialise..."
        time.sleep(1)

    print "Arming motors"
    # Copter should arm in GUIDED mode
    vehicle.mode    = VehicleMode("GUIDED")
    vehicle.armed   = True

    # Confirm vehicle armed before attempting to take off
    while not vehicle.armed:
        print " Waiting for arming..."
        time.sleep(1)
```

Then up your throttle using RC overrides to actually take off.

jainakshay91, 2016-04-19:
Yeah well I can do that, that was my initial thought. But since I am developing my own RC controller, I wanted to just try and pass the PWMs based on stick positions to get the drone to arm. I might just go the conventional way if a solution does not come up. 

Any ideas as to what could be causing the arm procedure to halt midway with "Initialising APM.." message.

hamishwillee, 2016-04-19:
YOu can't take off without explicitly arming AFAIK. Initialising must complete before you try to arm. Use the code above before your other code :-)

jainakshay91, 2016-04-19:
Yeah I will do that for now. Thanks :)