dronekit/dronekit-python #618: Drone does not arm using dronekit python for indoor operations

jainakshay91, 2016-04-22:
Background: I am looking to use my drone for indoor operations. Since I cannot get a good GPS signal indoors, I disable GPS check in the pre-arm checks by going into Mission planner and:
1. Unchecking it from arming checks list in the standard parameters
2. Setting AHRS_GPS_USE to 0 and Arming check to -9 in the full parameters list.

Problem: Now when I try to arm the drone using the conventional method as shown in the dronekit python documentation it shows: "Prearm: Need 3D fix". However, if I arm using my RC controller, it arms.  This is somehow pointing to the fact that somehow dronekit-python still performs a GPS 3D fix check even if we set it in the mission planner to ignore the GPS check. I am not sure if this is the case, but my experiments are pointing towards this fact. Hence, can somebody kindly provide some insight into this matter. I appreciate all of your time and help in advance. 

Versions of the platform in use --
OS: Linux 3.10.80-135
ArduCopter Version: 3.3.3
dronekit python version: 2.0

Thanks 
Best 
Akshay

jainakshay91, 2016-04-22:
Update: 
1. if I arm it in guided mode: It throws the "Prearm: Need 3D fix"
2. if I arm it in stabilize mode: it gets stuck at "Initializing APM..." 

If 3D fix is a hard set requirement, then I would suggest that a feature be added so that one can arm it even without 3D fix, as that is essential for indoor operations.

hamishwillee, 2016-04-24:
@jainakshay91 What is required to ARM is dependent on the autopilot - DroneKit doesn't have any ability to control that. What you can do for indoor use is disable the ARMING checks using a parameter. You will also have to change any code which checks on `is_armable` since that hard codes to check for GPS etc.

ArduPilot currently has only limited support for indoor use. Discussion of this should take place on the discussion boards: http://discuss.ardupilot.org/

closed 2016-04-24 by hamishwillee

squilter, 2016-04-25:
The relevant ardupilot issue is ArduPilot/ardupilot#2790

Guohao91, 2020-01-17:
@jainakshay91 I like your indoor projects idea.
[image]
my project is gonna to guide copter in underground (mining plant, no GPS signal at all), I've managed to disable the ARMing checks, but still my copter working with something wrong.
Have you figured out "Prearm: Need 3D fix" problem?
Looking forward your reply and further help! Thanks in advance!

leonardocfor, 2020-01-18:
@jainakshay91 I believe ArduPilot is not devised for indoor navigation by default and you will require to modify its source code to allow this. However, this is an issue that has been discussed in the past, as you can check here: https://discuss.ardupilot.org/search?q=indoor . Also, a quick search on google gave me these links:

https://ardupilot.org/copter/docs/indoor-flying.html
https://ardupilot.org/copter/docs/common-vicon-for-nongps-navigation.html
https://ardupilot.org/copter/docs/common-pozyx.html

Hope it helps

leonardocfor, 2020-01-18:
@jainakshay91 I see you are at UPC Barcelona, is that correct? If so, perhaps I can put you in contact with my former Ph.D advisor and his team who perhaps can help you with this issue