dronekit/dronekit-python #620: Can only control yaw from RC after executing dronekit commands

abearman, 2016-04-28:
I'm working on a project where it's useful to be able to control the drone from both dronekit and the remote controller (e.g., I'm running a script using the Python API, but want to make small adjustments with the RC). Is this supposed to be supported functionality? I've noticed that I'm only able to control **yaw** from the controller, but nothing else (the pitch, roll, and throttle sticks have no effect). This does not seem like behavior that is intended.

ivanovishx, 2016-06-04:
I'm doing something similar but using an external micro controller(aka arduino) communicating with mavlink to the Drone and and reading PPM receiver to overwrite RCchannels with mavlink, is that useful for you ?

Travis-Taylor, 2016-08-05:
I am having a similar issue. My project was implementing a joystick controller to send RC overrides to the pixhawk, but the RC controller could still send commands to change modes - none of its channels 1-4 (roll, pitch, throttle, yaw) were executed by the pixhawk. However, even when I removed the overrides functionality (and the joystick controller) from the project, the RC controller continued to be ignored for sending attitude commands over channels 1-4. Its mode changes are heard, but nothing else. Is this something to do with dronekit as a whole? If it somehow disables the RC commands from the RC controller for the duration of the dronekit script....

Anyway, @abearman, did you find a fix to your issue? Or does anyone else have any experience with this issue?

hamishwillee, 2016-08-07:
@Travis-Taylor In **guided mode** RC commands from the controller are disabled _by the autopilot_ (and scripts are run in guided mode). This is an autopilot thing (and really the only sensible approach) not a dronekit thing.

hamishwillee, 2016-08-07:
@abearman As above, this is an autopilot thing. My understanding is that RC input will be stopped in guided mode. 
I have not tried it, but I think you can still read the RC input from the controller, so you could manually mix it in with your override signal as needed. Does that help?

Travis-Taylor, 2016-08-08:
@hamishwillee Yes, it acts in accordance with that behavior, of course. But the autopilot still responds to mode changes (for example, to and from **loiter** and **guided** modes), and while in **loiter** mode the copter should still respond to RC attitude input. I certainly don't want the controller to act on the RC controls during **guided** mode, but if I switch to **loiter** mode I want the RC commands to continue dictating the attitude of the copter

I say all this, but in fact the behavior I was observing is no longer taking place. To explain, I was trying to switch from manual control in **loiter** mode and then, at a certain point, switch to **guided** and execute a simple_goto(), then back to **loiter**. However, during this it either would not change to **guided** at the appropriate time, or would not return control back to **loiter** mode. However, once I properly removed all traces of the channels.overrides functionality from the program (controlling the copter strictly by RC controller), the switching behavior worked fine. Surprisingly well, in fact, considering the difficulty I was having in earlier testing.

Anyway, thanks for your quick response. Hopefully my problem will help someone else at some point.

hamishwillee, 2016-08-08:
@Travis-Taylor So yes, switching modes should not IMO be affected by RC override behaviour. I think it most likely that this is due to behaviour in the autopilot. Your best bet is to ask about this in the ardupilot copter forums.