dronekit/dronekit-python #622: Support setting yaw rate and yaw acceleration

waTeim, 2016-05-05:
This feature is needed to preserve a smooth feel in response to target angle change while recording video.  Without it, the scene has a "jerky" feel and the subject move across the field of view instead of staying centered.  Importantly in contrast to upstream discussion that appear to talk about a global setting, this would need to be settable per command.

Some documentation from the various examples in particular; currently this appears to be unsupported but the example documentation might be old and the issues are closed.

### Ardu [documenation](http://ardupilot.org/copter/docs/common-mavlink-mission-command-messages-mav_cmd.html#mav-cmd-condition-yaw) of MAV_CMD_CONDITION_YAW
`The parameters allow you to specify whether the target direction is absolute or relative to the current yaw direction. If the direction is relative you can also (separately) specify whether the value is added or subtracted from the current heading (note that the vehicle will always turn in direction that most quickly gets it to the new target heading regardless of the param3 value).`

`We don’t support controlling the yaw rate, so the param2 value (Sec) is ignored.`

## Dronkekit [examples](http://python.dronekit.io/examples/guided-set-speed-yaw-demo.html)   

### send_global_velocity

        velocity_z, # Z velocity in NED frame in m/s
        0, 0, 0, # afx, afy, afz acceleration (not supported yet, ignored in GCS_Mavlink)
        0, 0)    # yaw, yaw_rate (not supported yet, ignored in GCS_Mavlink)

### condition_yaw

        heading,    # param 1, yaw in degrees
        0,          # param 2, yaw speed deg/s
        1,          # param 3, direction -1 ccw, 1 cw

## Ardu issues 
This is apparently a requested feature upstream. includes ArduPilot/ardupilot#1635, ArduPilot/ardupilot#1758

hamishwillee, 2016-05-16:
Neither of these are currently supported in ArduPilot (ie Copter 3.3). This is really an ardupilot question, not a dronekit question. I've asked for the right issues to track for this functionality, and will post them if that information is supplied to me. Closing!

closed 2016-05-16 by hamishwillee

hamishwillee, 2016-05-16:
Not a perfect match, but see https://github.com/ArduPilot/ardupilot/issues/1640

waTeim, 2016-05-16:
Yea not a great fit, but have to start somewhere.  It's remarkable such a large limitation has been left unaddressed for so long.

hamishwillee, 2016-05-17:
@waTeim I guess it hasn't been a limitation for others, or they've found other workarounds. If the requirement won't meet your need, either add a comment or add a new issue (whichever makes most sense in terms of a clear requirement spec)