dronekit/dronekit-python #623: Pixhawk and Raspberry in automode to perform image recognition

n0rmanUK, 2016-05-10:
Hi everyone!

I have been solving my problems with forum, but now I am stuck and I need to post for help.

I am undertaking a project with an aim to perform an autonomous mission with a drone that has to recognise a target to drop a payload. I would like to be connected to Pixhawk via the mission planner on the computer and have an RP connected as an additional tool to the Pixhawk. Once, the copter gets to the desired WP I would like to perform image recognition and send a signal to Pixhawk to deploy a payload. 

I know how to connect the two devices and I am aware of the DroneKit etc. My problem is: how do I hold an autonomous mission for purpose of image recognition?

Can I switch to loiter mode, perform what I want on the RP and then switch it back to auto, so it continues with a mission? I read the pixhawk will then restart the mission.

Is there a way of doing it without running the groundstation from the raspberry pi ?

hamishwillee, 2016-05-11:
Hi @n0rmanUK 

Best place for this sort of question is possibly the new discuss forums: https://discuss.dronekit.io/

> Can I switch to loiter mode, perform what I want on the RP and then switch it back to auto, so it continues with a mission? I read the pixhawk will then restart the mission.

Yes. You should perform the RP code in GUIDED mode (not Loiter). The pixhawk will either resume or restart the mission when you switch back to AUTO depending on the vehicle parameters (can't remember which one - look it up). Whether you restart, resume, or load a completely different mission is up to you.

In addition to sending commands in guided mode, you can in theory also dynamically create a mission in AUTO mode and upload the modified mission when needed. 

> Is there a way of doing it without running the groundstation from the raspberry pi ?

Probably not. You can't run your control script on the pixhawk itself so the RaPi is the only computing platform you have on board. While you can run the script from a PC say and communicate with the Pixhawk over a telemetry link, that is usually too high-latency (slow) for image processing work.

What are you envisaging?

ARPIT355, 2018-03-11:
how to configure pixhawk if i am not using RC and just using Raspberry pi....means can we do calibration using usb connected to pixhawk?