dronekit/dronekit-python #625: Moving precisely without GPS

alexblack, 2016-05-18:
I'd like to be able to move and position the drone without GPS, I'm working on a visual positioning system that uses the camera to detect a target from which position can be determined.

I'm using `set_position_target_local_ned_encode` on `vehicle.message_factory` to move the drone, and its generally working, but, the movements are not very precise, for example I attempted to move the drone 10cm, and it moved 48cm.

Questions:
1. Does `set_position_target_local_ned_encode` work without the use of GPS?
2. How should `set_position_target_local_ned_encode` be used for precise movements?

Currently I'm calling `set_position_target_local_ned_encode` every 100ms with a new velocity, on a curve, from 0 upto a max velocity and back to 0, where the integral of the curve is the distance I want to travel.

alexblack, 2016-05-18:
In particular altitude control seems poor, often it kinda of jumps up and down, could this be related to: http://diydrones.com/forum/topics/althold-inaccurate-altitude-estimation-and-motor-pulsing?xg_source=activity&id=705844%3ATopic%3A2170652&page=3#comments

hamishwillee, 2016-05-18:
@alexblack These are actually autopilot-specific questions, and you should raise them on the ardupilot.org discussion boards.

That said, in the discussions I've seen, ArduPilot has minimal support for guidance without GPS - there is some limited support when using an optical flow sensor (something to search on). More support is planned. 

Developers currently seem to mostly use rc overrides to control the vehicle.

closed 2016-05-18 by hamishwillee

alexblack, 2016-05-18:
hi @hamishwillee gotcha, thanks.. Hmm, just to be clear, I am looking specifically for how to ask the drone to attempt to move (x,y,z) relative to where it is: so it should fire the motors in such a way that it will likely end up where I asked it to without taking any location data as input.

hamishwillee, 2016-05-18:
Yes :-) http://python.dronekit.io/examples/channel_overrides.html

We strongly don't recommend using RC overrides for anything other than joystick simulation, but in this case I don't **believe** there are any reliable alternatives.

alexblack, 2016-05-18:
@hamishwillee ah cool, thanks I'd never heard of that.. So saying it back to you so I understand, it sounds like you're saying that `set_position_target_local_ned` relies on GPS, and the only way to actually tell the drone to make a movement (without GPS) is by using the RC overrides?

Whats the general concept there?  Is it similar to flying the drone using the controller without GPS? So if we wanted to move forward we'd use the RC channels to pitch the drone for some period of time?

hamishwillee, 2016-05-18:
@alexblack I am not an expert on the autopilot interpretation of each of those commands, which is why I pushed you to the discussion forums. That said, my understanding is:

`set_position_target_local_ned` relies [on EKF](http://ardupilot.org/dev/docs/extended-kalman-filter.html#extended-kalman-filter) which calculates the position based on a number of sources, including GPS. I believe this is largely tuned such that yes, the system is reliant on GPS. 

I know that there is limited support for using the optical flow instead of GPS - Randy Mackay has answered questions about that so search on discussion boards. 

> Whats the general concept there? Is it similar to flying the drone using the controller without GPS? So if we wanted to move forward we'd use the RC channels to pitch the drone for some period of time?

Yes :-) Basically the RC overrides tell the controller that some other throttle value (say) was sent from the controller than was actually sent.  Effectively your script pretends to be the controller.

> I think essentially I'm looking to make small movements of the drone using dead reckoning, and then I'll correct based on location data I pull from the camera.

Yes exactly. 

Again, check out the discussion on optical flow camera if you have one on your vehicle. If that doesn't do what you need then the above option is the one that most users have followed. Randy is the expert and is likely to answer questions on this if you post them in the [discussion boards](http://discuss.ardupilot.org/), especially if you do a due-diligence search first. Sorry I can't be more canonical - I've been trying to encourage Randy to help write an article on this, but we both have many other priorities.

alexblack, 2016-05-18:
Thats a big help @hamishwillee - thanks!

alexblack, 2016-05-25:
> set_position_target_local_ned relies on EKF which calculates the position based on a number of sources, including GPS. I believe this is largely tuned such that yes, the system is reliant on GPS.

I think there are actually two separate questions here:

1. If GPS is present, is `set_position_target_local_ned` the most accurate way to get the drone to make precise movements

2. If there is no GPS signal, how can the drone be controlled? 

You've answered #2 clearly.  What about #1, is it as clear?  We're determine our location using POSE estimates from images, and would then like to move the drone a bit closer to our target based on that.

ArathornII, 2017-08-03:
I've been experimenting with other pyMAVlink commands (not RC channel overrides) that would theoretically allow you control the velocity of the UAV by controlling its pitch and roll.

ArathornII, 2017-08-03:
And to be clear, I am also looking at how to accurately control my UAV in an indoor, GPS-denied environment.

sabrinarosecarroll, 2020-07-01:
@jacobfeldgoise Did you have any success using pyMAVlink commands to control the velocity of the UAV? 
I am looking to achieve a similar task of accurately controlling a UAV in an indoor, GPS-denied environment. I am hoping to use ArUco Tags for precise localization, but I haven't been able to get there just using GUIDED_NOGPS mode.

jacobfeldgoise, 2020-07-01:
I couldn't get it to work, if I remember correctly.

bhaskar-glitch, 2021-09-18:
> I've been experimenting with other pyMAVlink commands (not RC channel overrides) that would theoretically allow you control the velocity of the UAV by controlling its pitch and roll.

can you please explain which commands of pymavlink you used...
I need your help

angela804, 2022-07-12:
I even have the same issue right now and looking, I don't think there are any supports for non-gps environment movement. Or maybe I'm wrong