dronekit/dronekit-python #626: Connect() to DroneKit-SITL broken on OSX

a7scenario, 2016-05-23:
I am running dronekit-sitl on OSX El Capitan and can successfully connect to the SITL using APM Planner on tcp port 5760.  When I run the example code (http://python.dronekit.io/examples/vehicle_state.html) with the startup string "python test.py --connect tcp:127.0.0.1:5760", I see within the dronekit-sitl window that it detected a connection request (New connection on serial port 0 Closed connection on serial port 0), however, I get this error from the DroneKit example and an immediate timeout (see below).  

Using MAVLink 1.0
Connecting to vehicle on: tcp:127.0.0.1:5760
Traceback (most recent call last):
  File "test.py", line 33, in <module>
    vehicle = connect(connection_string, wait_ready=True)
  File "/Library/Python/2.7/site-packages/dronekit/__init__.py", line 2813, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/Library/Python/2.7/site-packages/dronekit/__init__.py", line 2086, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.

I am running the following versions:
dronekit (2.1.0b1)              - Python API for MAVLink enabled Drones
  INSTALLED: 2.5.0
  LATEST:    2.1.0b1
dronekit_sitl (3.0.1)

a7scenario, 2016-05-23:
update:

I uninstalled pymavlink 2.0.0 and re-installed pymavlink 1.1.69 and it resolved the issue.  The error above was occurring within pymavlink.py (2.0.0) on line 84, "newsendfn() got an unexpected keyword argument force_mavlink1"

tridge, 2016-05-23:
this is a bug in dronekit-python. It really should not be overriding an internal function in pymavlink like that.
The quick and dirty fix is to add the force_mavlink1 parameters to newsendfn(). 
A better fix is to add a set_target_system() API to pymavlink to implement this internally.

hamishwillee, 2016-05-23:
@peterbarker Is this something you can look at as per tridge comment?

tridge, 2016-05-23:
@peterbarker has suggested using **kwargs, which is indeed a better fix

peterbarker, 2016-05-23:
@hamishwillee Yep, I'll be working on this today.

hamishwillee, 2016-05-23:
Excellent, my work is done :-)

prat96, 2016-05-27:
Still having this issue on OSX. thanks @a7scenario for the temporary bugfix.

prat96, 2016-05-30:
I am getting the exact same error on Ubuntu 16.04 LTS! No idea how to fix this :/

rtfswr, 2016-05-30:
@a7scenario thanks, I removed pymavlink 2.0.0, installed pymavlink 1.1.69 and it works.

magnes919, 2017-04-19:
@prat96 I am working on Ubuntu 16.04 as well. Could you walk me through what you did to downgrade MAVProxy. Thanks.